slic3r/Slic3r #2471: spinning beach ball on open STL

abraren, 2015-01-01:
When I open an stl file Slic3r becomes unresponsive. Cura can one that stl file.
Slic3r is version 1.2.4
OS is MacOSX 10.9.5 Build 13F34
STL was exported from Inventor Fusion 576.1.0
[image]

abraren, 2015-01-01:
This is the sample file:
Sampling process 43385 for 3 seconds with 1 millisecond of run time between samples
Sampling completed, processing symbols...
Analysis of sampling slic3r (pid 43385) every 1 millisecond
Process:         slic3r [43385]
Path:            /Applications/CNC/Slicer/Slic3r.app/Contents/MacOS/slic3r
Load Address:    0x1000
Identifier:      slic3r
Version:         Slic3r 1.2.4.91 (1.2.4.91)
Code Type:       X86
Parent Process:  launchd [184]

Date/Time:       2015-01-01 16:19:46.167 +0100
OS Version:      Mac OS X 10.9.5 (13F34)
Report Version:  7

Call graph:
    2233 Thread_2432166   DispatchQueue_1: com.apple.main-thread  (serial)
    + 2233 ???  (in slic3r)  load address 0x1000 + 0x35a5  [0x45a5]
    +   2233 ???  (in slic3r)  load address 0x1000 + 0x5d2b  [0x6d2b]
    +     2233 perl_run  (in libperl.dylib) + 366  [0x7542e]
    +       2233 Perl_runops_standard  (in libperl.dylib) + 22  [0xe5af6]
    +         2233 Perl_pp_entersub  (in libperl.dylib) + 2294  [0xe74a6]
    +           2233 XS_Wx__App_MainLoop(interpreter*, cv*)  (in Wx.bundle) + 135  [0x5643717]
    +             2233 wxPliApp::MainLoop()  (in Wx.bundle) + 46  [0x565d772]
    +               2233 wxAppConsoleBase::MainLoop()  (in libwx_baseu-2.9.3.0.0.dylib) + 97  [0x6054285]
    +                 2233 wxCFEventLoop::Run()  (in libwx_baseu-2.9.3.0.0.dylib) + 64  [0x612a0ba]
    +                   2233 wxGUIEventLoop::DoRun()  (in libwx_osx_cocoau_core-2.9.3.0.0.dylib) + 51  [0x5bb6c17]
    +                     2233 -[NSApplication run]  (in AppKit) + 727  [0x96d1a15c]
    +                       2233 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:]  (in AppKit) + 119  [0x96d27870]
    +                         2233 _DPSNextEvent  (in AppKit) + 1602  [0x96d28349]
    +                           2233 _BlockUntilNextEventMatchingListInModeWithFilter  (in HIToolbox) + 92  [0x969b36bd]
    +                             2233 ReceiveNextEventCommon  (in HIToolbox) + 526  [0x969b38e2]
    +                               2233 RunCurrentEventLoopInMode  (in HIToolbox) + 259  [0x969b3b5d]
    +                                 2233 CFRunLoopRunInMode  (in CoreFoundation) + 123  [0x982f084b]
    +                                   2233 CFRunLoopRunSpecific  (in CoreFoundation) + 394  [0x982f09ea]
    +                                     2233 __CFRunLoopRun  (in CoreFoundation) + 2130  [0x982f14b2]
    +                                       2233 __CFRunLoopDoSource1  (in CoreFoundation) + 523  [0x98300d7b]
    +                                         2233 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE1_PERFORM_FUNCTION__  (in CoreFoundation) + 53  [0x98300e15]
    +                                           2233 __CFMessagePortPerform  (in CoreFoundation) + 832  [0x98383ba0]
    +                                             2233 CoreDragMessageHandler  (in HIServices) + 960  [0x99844eaf]
    +                                               2233 DoDropMessage  (in HIServices) + 66  [0x99845f70]
    +                                                 2233 DoMultipartDropMessage  (in HIServices) + 346  [0x998462bf]
    +                                                   2233 NSCoreDragReceiveMessageProc  (in AppKit) + 655  [0x9705134b]
    +                                                     2233 wxOSX_performDragOperation(objc_object*, objc_selector*, objc_object*)  (in libwx_osx_cocoau_core-2.9.3.0.0.dylib) + 52  [0x5bd3744]
    +                                                       2233 wxWidgetCocoaImpl::performDragOperation(void*, NSView*, void*)  (in libwx_osx_cocoau_core-2.9.3.0.0.dylib) + 314  [0x5bd1d4a]
    +                                                         2233 wxFileDropTarget::OnData(int, int, wxDragResult)  (in libwx_osx_cocoau_core-2.9.3.0.0.dylib) + 55  [0x5c06a5b]
    +                                                           2233 wxPliFileDropTarget::OnDropFiles(int, int, wxArrayString const&)  (in DND.bundle) + 289  [0x53de6d7]
    +                                                             2233 wxPliVirtualCallback_CallCallback(interpreter*, wxPliVirtualCallback const*, long, char const*, ...)  (in Wx.bundle) + 51  [0x5651ea3]
    +                                                               2233 wxPliVirtualCallback::CallCallback(interpreter*, long, char const*, char*&) const  (in Wx.bundle) + 225  [0x565b619]
    +                                                                 2233 Perl_call_sv  (in libperl.dylib) + 525  [0x74bdd]
    +                                                                   2233 Perl_runops_standard  (in libperl.dylib) + 22  [0xe5af6]
    +                                                                     2233 Perl_pp_entertry  (in libperl.dylib) + 61  [0x12056d]
    +                                                                       2233 S_docatch  (in libperl.dylib) + 102  [0x120496]
    +                                                                         2233 Perl_runops_standard  (in libperl.dylib) + 22  [0xe5af6]
    +                                                                           2233 Perl_pp_entersub  (in libperl.dylib) + 2294  [0xe74a6]
    +                                                                             2233 XS_Slic3r__TriangleMesh_vertices(interpreter*, cv*)  (in XS.bundle) + 1939  [0x505ac93]
[... cut ...]
0x9c237000 - 0x9c238fff  liblangid.dylib (117) <F18F76C6-7E4B-34AD-AE81-C1C031BF2F7D> /usr/lib/liblangid.dylib
0x9c239000 - 0x9c239ffd  com.apple.audio.units.AudioUnit (1.10 - 1.10) <9515158F-3A33-39CF-AD5A-201C2B121F33> /System/Library/Frameworks/AudioUnit.framework/Versions/A/AudioUnit
0x9c23a000 - 0x9c27aff7  com.apple.bom (14.0 - 193.1) <FFF1C8E5-41FF-357B-8681-69B21DCED2E4> /System/Library/PrivateFrameworks/Bom.framework/Versions/A/Bom
Sample analysis of process 43385 written to file /dev/stdout

Laura3, 2015-01-03:
Could you maybe edit your post and put some code boxes around that? Saves for lots of scrolling.
Here's how: https://help.github.com/articles/github-flavored-markdown/#fenced-code-blocks

alexrj, 2015-01-04:
@abraren, can you post the STL file and configuration (File -> Export config...)? They are needed for reproducing the issue.

abraren, 2015-01-05:
config.ini:
```
# generated by Slic3r 1.2.4 on Mon Jan  5 20:34:29 2015
avoid_crossing_perimeters = 1
bed_shape = 0x0,184x0,184x188,0x188
bed_temperature = 70
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 60
brim_width = 0
complete_objects = 0
cooling = 1
default_acceleration = 0
disable_fan_first_layers = 1
dont_support_bridges = 1
duplicate_distance = 6
end_gcode = M104 S0 ; turn off temperature\nM140 S20\nG28 X0  ; home X axis\nM84     ; disable motors\nM106 S129
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 70%
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0
fan_always_on = 0
fan_below_layer_time = 60
filament_diameter = 3
fill_angle = 45
fill_density = 30%
fill_pattern = honeycomb
first_layer_acceleration = 0
first_layer_bed_temperature = 70
first_layer_extrusion_width = 200%
first_layer_height = 150%
first_layer_speed = 30%
first_layer_temperature = 200
gap_fill_speed = 20
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
infill_speed = 60
interface_shells = 0
layer_gcode = 
layer_height = 0.36
max_fan_speed = 100
min_fan_speed = 35
min_print_speed = 10
min_skirt_length = 0
notes = 
nozzle_diameter = 0.4
octoprint_apikey = 
octoprint_host = 
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 25
perimeters = 3
post_process = 
pressure_advance = 0
raft_layers = 0
resolution = 0
retract_before_travel = 2
retract_layer_change = 0
retract_length = 1
retract_length_toolchange = 10
retract_lift = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 30
seam_position = aligned
skirt_distance = 10
skirt_height = 1
skirts = 2
slowdown_below_layer_time = 30
small_perimeter_speed = 25
solid_infill_below_area = 70
solid_infill_every_layers = 4
solid_infill_extruder = 1
solid_infill_extrusion_width = 0
solid_infill_speed = 60
spiral_vase = 0
standby_temperature_delta = -5
start_gcode = G28 ; home all axes\nG1 Z5 F5000 ; lift nozzle\nM106 S129
support_material = 1
support_material_angle = 37
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 0.5
support_material_interface_speed = 100%
support_material_pattern = rectilinear
support_material_spacing = 2.5
support_material_speed = 60
support_material_threshold = 0
temperature = 200
thin_walls = 1
threads = 4
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 50
top_solid_layers = 3
travel_speed = 80
use_firmware_retraction = 0
use_relative_e_distances = 0
vibration_limit = 0
wipe = 1
xy_size_compensation = 0
z_offset = 0
```

abraren, 2015-01-05:
Not sure how to post the STL. I send it as attachment.
The config.ini I posted as comment.

HTH 
Arfst

alexrj, 2015-01-07:
Mail attachments don't work with GitHub. Can you upload it somewhere and put a link here? Dropbox, wetransfer, whatever file hosting service is fine. Thank you.

abraren, 2015-01-07:
STL File is here:
http://www.file-upload.net/download-10107832/asl5.stl.html

alexrj, 2015-01-08:
Thank you. I confirm Slic3r fails when parsing such file, but that's caused by the file being invalid as it contains multiple "solid" definitions.
Whether this can lead to parsing errors depends on how the parser is written. I'll try to make the Slic3r STL parser more tolerant.

alexrj, 2015-06-01:
I fixed this. Thank you for reporting.

closed 2015-06-01 by alexrj