slic3r/Slic3r #2472: Feature request: auto eject of SD cards

zapta, 2015-01-01:
Like many I am using SD cards to transfer the gcode generated by Slic3r to my 3D printer and going through the cycle of modify the design, export STL, load STL, write gcode and eject SD card is tedious.

It would be very useful to have in Slic3r an auto-eject check configuration box that automatically ejects the SD card after writing the gcode file to it.

simonkuehling, 2015-01-01:
I think this may require quite platform-and maybe even distribution-specific code for each operating system to implement in Slic3r itself..

Did you consider using a post-processing script for that task already? In linux and OSX this could be a one-liner i think - i am not sure how easy it is for windows though

zapta, 2015-01-02:
I am using Mac OSX, I managed to setup an 'eject' button in the top desktop bar but it doesn't work for SDs. Yes, it may be platform dependent but very useful consider how popular SD cards are. For platforms that do not support it it can be disabled. With the right abstraction it should not be that bad I think. It can be as simple as dispatching a shell command from Slic3r.

alexrj, 2015-01-02:
Detection of SD cards might require some hacks... This is something I've been considering myself too. We need to investigate how to do that on Windows.

zapta, 2015-01-02:
Does this help?

http://msdn.microsoft.com/en-us/library/windows/desktop/aa364939%28v=vs.85%29.aspx

lordofhyphens, 2016-07-11:
I think this is better solved as a post process script myself, as there are as many ways to unmount a file system as there are OSes. Keeping it in Slic3r itself would also start to necessitate permission escalations on some OS configurations. 

This is also not my use case

bubnikv, 2016-07-11:
Cura (the original gui by Daid) implements SD card detection very nicely as
far as I can judge it on Windows. Cura also implements multi platform SD
card eject, but it is disabled by default. I suppose there is some reason
for it being disabled. On Windows, cura uses an external 3rd party tool for
eject.

dbuezas, 2023-12-05:
Since the suggested post-processing script was trickier than I expected, posting it here in case others land here:

```bash
(
sleep 1 # so orca starts saving first
diskutil eject /dev/disk2
say "Ejected" # mac specific (I think)
) >/dev/null 2>&1 &
```

Put in a file, `chmode +x`, put script's path in the post-processing field