slic3r/Slic3r #2473: Feature request: Auto reload of STL files

zapta, 2015-01-01:
Like many I am using SD cards to transfer the gcode generated by Slic3r to my 3D printer and going through the cycle of modify the design, export STL, load STL, write gcode and eject SD card is tedious.

It would be very useful to have in Slic3r an option to refresh the currently loaded STL files. Either by a reload button or by watching changes in the file system.

Piesi, 2015-01-01:
thanks, this currently is the feature I'm missing most!

Laura3, 2015-01-02:
You could make a batch file or shell script to run slic3r with your settings that takes the stl as an argument, puts the gcode on the sd card and autounmounts it.

Simply run it and it does all the things you wanted.
It will take some manual reading, googling, but it's not that hard.

zapta, 2015-01-02:
Thanks Alex. I want to run Slic3r interactivity (reviewing the model,
setting printing parameters, etc), but to simplify the mechanical parts.

I looked around and figured out how to mount/unmount the SD on my mac OSX
machine. If only I could tell slic3r to run this command automatically when
it writes out the gcode file to that disk.

~$ diskutil list

/dev/disk0

   #:                       TYPE NAME                    SIZE
IDENTIFIER

   0:      GUID_partition_scheme                        *121.3 GB   disk0

   1:                        EFI EFI                     209.7 MB   disk0s1

   2:          Apple_CoreStorage                         120.5 GB   disk0s2

   3:                 Apple_Boot Recovery HD             650.0 MB   disk0s3

/dev/disk1

   #:                       TYPE NAME                    SIZE
IDENTIFIER

   0:                  Apple_HFS MacintoshHD            *120.1 GB   disk1

/dev/disk3

   #:                       TYPE NAME                    SIZE
IDENTIFIER

   0:     FDisk_partition_scheme                        *7.9 GB     disk3

   1:                 DOS_FAT_32 3D PRINTER              7.9 GB     disk3s1

~$ diskutil unmountdisk /dev/disk3

Unmount of all volumes on disk3 was successful

~$ diskutil mountdisk /dev/disk3

Volume(s) mounted successfully

Laura3, 2015-01-02:
Put the command you want to run in a text file and rename the file extension to .sh
I don't know about macs, but on linux systems you have to set the file as 'allow file to be executed' under properties on the file. 

Then under slic3r's print settings tab, go to advanced, then under post processing script enter the path to your unmount.sh file
so for me that would be /home/user/Desktop/unmount.sh
Tadum! now it runs everytime you've finished exporting gcode.

zapta, 2015-01-03:
Yes, this worked. Thanks!  (found it under Output Options).

Is there a similar trick for pre processing script? This way I can delete
the old .gcode file so I don't need to confirm the 'overwrite' each time.

BTW, initially I had an extra space after the script name and it didn't
work. Looks to me the arg requires space trimming.

zapta, 2015-01-03:
on mac osx I am using this shell script. /Volumes/3D PRINTER is my 3D
printing SD card.

#!/bin/bash -x
diskutil eject "/Volumes/3D PRINTER"
if [ $? -eq 0 ]
then
  tput bel
else
  tput bel
  tput bel
  tput bel
fi

zapta, 2015-01-03:
... but I cannot hear the bell when it runs slic3r.

zapta, 2015-01-03:
... this script works better. Still no solution for the 'Replace'
confirmation.

#!/bin/bash -x

function ok() {
  afplay -v 0.1  /Users/user/util/scripts/ok.wav
}

function error() {
  afplay -v 0.1  /Users/user/util/scripts/error.wav
}

diskutil eject "/Volumes/3D PRINTER"
if [ $? -eq 0 ]
then
  ok
else
  error
fi

Laura3, 2015-01-03:
You can't just have spaces in the file name, you need to put a backslash in front of the spaces like this:
/home/user/Desktop/3D\ PRINTER
I generally avoid putting spaces in the names, or use an underscore instead.

zapta, 2015-01-04:
My comment was about a space after the file name. E.g. if the file name is
/bin/myscript.sh and I enter it into slc3r with an additional
(unintentional) space after it, slic3r does not make the extra space
visible and doesn't run the file.

it's common to trim leading and trailing spaces in fields before using
their values.

Tr

Laura3, 2015-01-04:
Right, so it might be nice to get that fixed..

rdomunky, 2015-06-30:
+1

My current workflow is to generate G-code in the Slic3R GUI, then send to OctoPrint from within Slic3r. When tweaking an STL and re-printing, OctoPrint sees the G-code update and refreshes accordingly. It would be great for Slic3r to do the same once the source STL changes on disk/Dropbox. Currently I need to delete and re-import the STL.

peter1960, 2015-06-30:
+1 on the option to reload the stl or detect/prompt to reload

alexrj, 2015-07-04:
I'm closing this one since it's a duplicate of #1466

closed 2015-07-04 by alexrj