slic3r/Slic3r #2474: Infill missing, preview doesn't show

Piesi, 2015-01-01:
Slic3r version: 1.2.4
OS: Ubuntu 12.04 64Bit

Expected result: Infill should be everywhere, gcode-preview should reflect the actual gcode.
Actual result: A part of the infill is missing in the actual gcode, but not in the Slic3r-preview.

Slic3r preview:
[image]

actual gcode (displayed in octoprint):
[image]

files (didn't found a way to attach to this issue):
https://www.dropbox.com/sh/qc9rtwu8h7oxyep/AACV_R1dOt3Qy7JVWS1N_mmsa?dl=0

a4jp-com, 2015-01-02:
How are you trying to print this? Do you have supports turned on?

[image]

Piesi, 2015-01-02:
it's rotated 90 degrees along the x-axis

a4jp-com, 2015-01-02:
I rotated it as you said.

[image]

fpiesik, 2015-01-02:
The thing is, that the slic3r preview doesn't show the missing infill. look at my screenshots, in the slic3r preview the infill is there, but in the actual gcode, there is some infill missing. Yes, I actually use a 0.8mm nozzle as it prints much faster and the resolution is still good enough for most of my print jobs.

a4jp-com, 2015-01-02:
Maybe Octoprint has the problem. Have you tried printing using another program? I would also have at least 2 perimeters if possible. Go into the advance settings for the time being and get some more. See if that makes a difference. It might not fix Slic3r but it might fix the G-code for printing.

fpiesik, 2015-01-03:
nope, it's not a problem with octoprint. here is a screenshot of the second layer made with pronterface:
[image]

With a 0.8mm nozzle there is no need for two perimeters, that speeds up the printing process significantly. I have made many good and fast prints this way. 

actually, the infill is only incorrect in a few layers which is not a very big problem in this case.

I think the main issue here is the indifference between the slic3rs gcode preview and the actual gcode.

a4jp-com, 2015-01-03:
That's okay but please change the settings to see if there is a bug.

fpiesik, 2015-01-03:
sorry, but I don't get the point of changing the paras to something i don't want. I have used many different parameters and never noticed this beahviour before. I provided everything needed to prove that under this circumstances there IS a bug. the gcode differs from what the preview shows and from what it should be...

a4jp-com, 2015-01-03:
Okay. If you can't try then I can't help you. Sorry.

alexrj, 2015-01-03:
@fpiesik, good catch there. I'm able to reproduce this issue, and the problem is not in the viewer since the missing segments are actually missing from G-code.

alexrj, 2015-01-03:
Ouch, I'm suddenly unable to reproduce this anymore... :(

fpiesik, 2015-01-03:
ah, good to hear that and how it can (not) be reproduced:)

alexrj, 2015-01-03:
Very tricky issue, combining two distinct minor bugs that I managed to isolate into regression tests. One of them is already fixed.

alexrj, 2015-05-22:
Fixed!

closed 2015-05-22 by alexrj

fpiesik, 2015-05-23:
Great!