slic3r/Slic3r #2475: corrupt stl after exporting cut model

walruses, 2015-01-01:
Version: 1.2.4 (64 bit)
OS: Win 7 64 bit

steps to reproduce:
For simplicity, we  can just use a simple 10x10x10 cube, although any model works:
https://www.dropbox.com/s/td98c1r1dvhx393/cube.stl

import the model > cut > z=5 > only keep bottom part > export .stl
(again any z value works)

here is the resulting stl:
https://www.dropbox.com/s/3szy0n4e4ar2aop/corrupt.stl

Attempting to load this file with Slic3r instantly force closes it. It is not possible to open the resulting stl in any CAD program (always causes a crash or error)

gege2b, 2015-01-02:
just tried on wheezy x64 and this worked fine
I'll test as soon as I reboot on win7

gege2b, 2015-01-02:
tested and confirmed under win 7 x64; it just crash

alexrj, 2015-01-08:
Hmm, I can't reproduce this on Win7 x64 with current git HEAD and with 1.2.4.

@walruses, @gege2b, are you using the compiled 1.2.4 package or are you running sources via git?

gege2b, 2015-01-09:
I use the 1.2.4 compiled windows binary
I just tested with the freshly downloaded  1.2.5 x64 and it crash again

here is a zip file containing my config (just snipped the octoprint stuff from it), the original STL and the cutted one

https://dl.dropboxusercontent.com/u/34934879/2475-slic3r-files.zip

note : I tested another STL just to see...
Here's the cutted STL : https://dl.dropboxusercontent.com/u/34934879/20150109-214933-dalek1_1.25_Larger_repaired_0.2-15.stl

Slic3r crashed again, but this time showing an error : 
[image]

If I look into the windows event log, there is one line per crash, anyway I'm not sure it could help
```
Nom du journal :Application
Source :       Application Error
Date :         09/01/2015 21:51:17
ID de l’événement :1000
Catégorie de la tâche :(100)
Niveau :       Erreur
Mots clés :    Classique
Utilisateur :  N/A
Ordinateur :   Gege-PC
Description :
Nom de l’application défaillante slic3r.exe, version : 1.2.5.35, horodatage : 0x503fe867
Nom du module défaillant : ntdll.dll, version : 6.1.7601.18247, horodatage : 0x521eaf24
Code d’exception : 0xc00000fd
Décalage d’erreur : 0x0000000000053443
ID du processus défaillant : 0x428
Heure de début de l’application défaillante : 0x01d02c4dfbb2443e
Chemin d’accès de l’application défaillante : C:\***\Slic3r\slic3r.exe
Chemin d’accès du module défaillant: C:\Windows\SYSTEM32\ntdll.dll
ID de rapport : 41344b85-9841-11e4-8240-60a44cb48f72
XML de l’événement :
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Level>2</Level>
    <Task>100</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2015-01-09T20:51:17.000000000Z" />
    <EventRecordID>9910</EventRecordID>
    <Channel>Application</Channel>
    <Computer>Gege-PC</Computer>
    <Security />
  </System>
  <EventData>
    <Data>slic3r.exe</Data>
    <Data>1.2.5.35</Data>
    <Data>503fe867</Data>
    <Data>ntdll.dll</Data>
    <Data>6.1.7601.18247</Data>
    <Data>521eaf24</Data>
    <Data>c00000fd</Data>
    <Data>0000000000053443</Data>
    <Data>428</Data>
    <Data>01d02c4dfbb2443e</Data>
    <Data>C:\***\Slic3r\slic3r.exe</Data>
    <Data>C:\Windows\SYSTEM32\ntdll.dll</Data>
    <Data>41344b85-9841-11e4-8240-60a44cb48f72</Data>
  </EventData>
</Event>
```

hope it helps

walruses, 2015-01-09:
I also used the 1.2.4 binary.
Just tried the 1.2.5 binary with somewhat different results. The good news is that it works about 8 times out of 10 that I tried it. However, When it doesn't work it just force-closes without any error message or entry in event viewer.
Here is my config: https://www.dropbox.com/s/o21piaq8t1saw08/Slic3r_config_bundle.ini

alexrj, 2015-02-01:
@walruses, what about 1.2.6?  Have we finally fixed this? I hope so, let me know...

alexrj, 2015-03-09:
@walruses, did you have a chance to test with 1.2.6?

gege2b, 2015-03-09:
Oh sorry, didn't saw this notification

Just tested with Win7 x64 with the precompiled 1.2.6-exp and this STL : https://dl.dropboxusercontent.com/u/34934879/darthVader_cleaned_fixed.stl

it doesn't crash, but the STL still seems invalid, Slic3r says it's empty

Here are the resulting STLs
https://dl.dropboxusercontent.com/u/34934879/20150309-160905-darthVader_cleaned_fixed_0.2-15.stl

https://dl.dropboxusercontent.com/u/34934879/20150309-161106-darthVader_cleaned_fixed_bottom_not_pivoted.stl
(for this one, I've unchecked the "rotate lower part" option)

Trying to open those STLs in MeshLab, I get the message "mesh contains 64 vertices with NAN coords and 0 degenerated faces." but nothing shows up

I've got a strange behaviour when doing this, I'll search for an already posted issue and eventually open one

walruses, 2015-03-09:
I'm getting similar results under 1.2.6, using the same darth vader model as gege2b. Under all parameters, the cut model produces an empty stl. However, under some (not all) z heights, attempting to cut the model will crash slic3r. The parameters i'm using are:
cut z:15.5, keep upper part only, rotate lower part upwards.

Source
slic3r.exe

Summary
Stopped working

Date
‎3/‎9/‎2015 3:55 PM

Status
Report sent

Description
Faulting Application Path:	D:\[redacted]\slic3r-mswin-x64-1-2-6-experimental\Slic3r\slic3r.exe

Problem signature
Problem Event Name:	APPCRASH
Application Name:	slic3r.exe
Application Version:	1.2.6.36
Application Timestamp:	503fe867
Fault Module Name:	ntdll.dll
Fault Module Version:	6.1.7601.18247
Fault Module Timestamp:	521eaf24
Exception Code:	c0000005
Exception Offset:	000000000005422c
OS Version:	6.1.7601.2.1.0.768.3
Locale ID:	1033
Additional Information 1:	51d9
Additional Information 2:	51d91d2cb25ee65434373073e18c9818
Additional Information 3:	3157
Additional Information 4:	3157bc41f8ebe8d4a36cf7104ed28d83

Extra information about the problem
Bucket ID:	123026398

alexrj, 2015-06-02:
@gege2b and @walruses, one more request... can you test with 1.2.7?  Several crashes and memory problems on Windows were fixed in that version, so I think it's worth testing this issue on 1.2.7.  Thank you in advance for your help.

alexrj, 2015-06-20:
@gege2b and @walruses, please let me know if Slic3r 1.2.9 fixes the issue. Thank you

gege2b, 2015-06-20:
seems fixed (in the freshly 1.2.9a) ! good job ^_^
again sorry, didn't see the notifications... that's why I just tested with 1.2.9a and not 1.2.8 or 1.2.7

alexrj, 2015-06-20:
Thank you @gege2b!

closed 2015-06-20 by alexrj