slic3r/Slic3r #2476: Contributors cut off in About dialog

blue-ice, 2015-01-01:
Slic3r version 1.1.7
Lubuntu Trusty 14.04 LTS

Any config file can be used. The expected result is that all contributor names should be visible in the About dialog. Instead, only a few can be viewed, and the rest are cut off. The window cannot be resized.

[image]

walruses, 2015-01-01:
Same issue in 1.2.4 on windows 7

alexrj, 2015-01-08:
Thank you! I increased the window height.

closed 2015-01-08 by alexrj

xoan, 2017-05-19:
This issue is happing again.

[image]

It is a low priority issue, and it will be probably fixed when GUI finishing porting from Perl to C++, so this comment it is only to taking this into account, and to look for a better way to set window size depending on content.