slic3r/Slic3r #2477: unideal bridge direction

walruses, 2015-01-01:
version 1.2.4 64 bit

model:https://www.dropbox.com/s/qxprwbjgljlvx0k/lego.stl
[image]
As you can see, slic3r attempts to bridge lengthwise. Although this technically works, if the bridge direction was rotated 90 degrees, the length of the bridge would be much shorter.

Ideally, there should be an option to override bridge direction.

harriv, 2015-01-01:
Ideally, Slic3r would automatically find the optimal solution :)

blue-ice, 2015-01-02:
Windows 7?

walruses, 2015-01-02:
I'm getting identical results in Win 7 and XUbuntu.

alexrj, 2015-01-05:
Good test case. The current logic should actually suggest the other direction: Slic3r chooses the direction that has most coverage and shortest max line length - in this case both directions have equal coverage, so it shouldn't be choosing this one. However coverage of angle 0° results slightly greater because of the way it is computed.
I think I should find a more robust algorithm for computing coverage/visibility.

alexrj, 2017-03-18:
Hm, I think I don't have the STL file anymore. I'll check on another computer.

alexrj, 2017-03-27:
[lego.stl.zip](https://github.com/alexrj/Slic3r/files/872073/lego.stl.zip)

alexrj, 2017-04-03:
This was an excellent test case. I improved the bridge detection code and also applied one more algorithm to handle bridges covering pillars. A regression test was added to the test suite.

closed 2017-04-03 by alexrj