slic3r/Slic3r #2478: Scroll wheel zoom direction in the 3d view is clearly wrong!

Lenbok, 2015-01-01:
Actually, I realise it's fairly arbitrary, but when using the scroll wheel to zoom in the 3d preview, the direction is inverted when compared to most other apps in the 3d printing toolchain, like: openscad, netfab, tatlin, blender, thingiverse 3d view, and the octoprint gcode viewer. With all of these, pushing the scroll wheel forward makes the 3d object bigger.  Please invert the scroll wheel direction for consistency with these apps.

Lenbok, 2015-01-01:
(BTW, I am using 1.2.4, which is awesome!!)

gege2b, 2015-01-02:
I agree, it's very disturbing ^_^

a4jp-com, 2015-01-02:
Or add an option to reverse it please ^^

alexrj, 2015-01-04:
@a4jp-com, it was you that asked to reverse it (#2420)... <facepalm>

alexrj, 2015-01-06:
Anyway, I reversed it again.

closed 2015-01-06 by alexrj