slic3r/Slic3r #2479: Feature: Ability to export individual split parts as STL

Lenbok, 2015-01-02:
Several times I have been sent a STL corresponding to a plate of parts to print. The split feature is great for rearranging them to better fit on the printer platform. However, often I want to save some individual parts (either to print separately, subdivide the plate, or further tweak the part). Currently the only option is to delete all the other parts of the split and then export the STL of the remaining part. Very tedious if you want to do this for multiple parts of the plate (as the original must be re-imported / split, etc).

Is it possible to have the right-click menu for an individual part contain an export STL option?

alexrj, 2015-01-02:
Good idea!

alexrj, 2015-01-03:
Done!

closed 2015-01-03 by alexrj