slic3r/Slic3r #2481: Octoprint integration: 500 error

PxT, 2015-01-02:
Slic3r 1.2.4
Mac OS 10.10.1

If I upload a very large gcode file (20MB) I get an error back:  
```Error while uploading to the OctoPrint server: 500 read timeout```

I can upload the same file manually with no error.

PxT, 2015-01-02:
[image]

alexrj, 2015-01-02:
What OctoPrint version?

PxT, 2015-01-02:
Octoprint Version: 1.1.1 (master branch)

alexrj, 2015-01-04:
I increased the timeout now. Works also with large files.

closed 2015-01-04 by alexrj

twproject, 2015-03-27:
I have the same problem in the devel branch
Version: 1.2.0-dev-687-g45bf398 (devel branch)

lordofhyphens, 2015-03-27:
That got fixed months ago, and the version number you give makes no sense.
Use the master branch.