slic3r/Slic3r #2482: Dragging the 3D View stops when hovering an object

gege2b, 2015-01-02:
Hi all
Commit 69da8b09997c58c3f47d1e676fe578f68cbb96e9
Debian Wheezy x64

When I import an object and drag the  3D view to turn around it, if the mouse pass over the object (when it highlights), sometimes the dragging stops

Well, its not a big issue, but its an annoying one (like the wheelzoom inverted ^_^)

a4jp-com, 2015-01-02:
This is kind of the same problem I have:

https://github.com/alexrj/Slic3r/issues/2454

gege2b, 2015-01-02:
Exact, I didn't found your issue before posting mine
I was speaking about rotating the view, but pan is also chaotic

I observed that the pan/rotation of the view is dropped as soon as the object is highlighted, and sometimes the object is highlighted even if the mouse isn't hover him.

alexrj, 2015-01-07:
Okay, I fixed this.

closed 2015-01-07 by alexrj