slic3r/Slic3r #2484: 1.2.4 current - top/bottom solid infill is always rectilinear

koghe, 2015-01-02:
Hi 
as the title say
solid infill is always rectilinear, in this case i think will be better a concentric path
to save vibration and speed, it make sense?

[image]

[image]

a4jp-com, 2015-01-02:
What OS are you using? Your icons look heaps better than on my PC version of Slic3r.

koghe, 2015-01-02:
Ubuntu 14.04 MBuntu OS Y icons theme, MediterraneanDarkest GTK, Mbuntu-Y-for-unity windows, with one of the last commit the save and delete icons in the expert mode look too small (i think its a theme problem and not slic3r)

[image]

how about the fill path?

MorganSandercock, 2015-01-04:
But you need extrusion crossing points to tie together the outside layer. If you had 3 top or bottom layers all set to concentric then you could get lots of little stacked rings that don't stick to each other within the layer. For most objects, you need to get these layers to stick very well and you don't care what the 2nd layer looks like.

For the interior fill less than 100% then concentric does make independent rings which aren't intended to stick to each other. This is probably not useful on most objects.
[image]

a4jp-com, 2015-01-04:
Being able to select an outside skinning layer and the type of pattern for that layer as an option or filter would be nice. If the fill for the skinning layer is 100% at the bottom, the sides of the concentric rings would touch and there wouldn't be a large gap but maybe the layer on the top surface needs a rectilinear layer under the selected pattern as Morgan points out. 

As there is no gap between the circles on the bottom it should be easy for the next layer to stick to the other layers but there might be problems depending on the infill amount when doing the top. Hopefully this idea fixes the problem and allows us to select a skinning layer pattern.

koghe, 2015-01-04:
@MorganSandercock, @a4jp-com is right, i can achieve the concentric layers by setting to 0 the tob/bottom layers and set the fill to 100%, but in this way you have all the layers at 100% infill, time decreased from 4.20 hours to about 3.15... 
An option to control the pattern and % infill will be nice.

alexrj, 2015-01-04:
Wait, there's a lot of confusion here.

For *external* solid layers, there's an option that lets you configure the infill pattern to use.
For *internal* solid layers, cross-hatched rectilinear is always used.

koghe, 2015-01-04:
Hi yes i know is not a bug, more a feature request, having an option to change the internal layers infill was just a suggestion

lordofhyphens, 2016-07-12:
Is there a reason why the interior solid infill has to be rectilinear? I can see some reason for it (although remember it's 100%) re: vibration and motion.