slic3r/Slic3r #2485: X/Y differences when "Detect thin walls" is disabled

xoan, 2015-01-02:
OpenSCAD model: http://hastebin.com/xiwakacupu.erl (walls are 0.5mm; it is only a testcase --print it with "Detect thin walls" disabled has no sense)
STL file: http://hastebin.com/nametufuci.vbs
Slic3r config: http://hastebin.com/edegatotis.hs

1. Load the same model twice
2. Rotate one of them 90º

When "Detect thin walls" option is enabled, G-code generated file is as expected:
[image]

But with this option disabled, there are significant differences when the geometry is in X axis and when it is in Y axis:
[image]

Also some other strange things happen when this walls are 45º:
[image]

a4jp-com, 2015-01-03:
When I was printing a ring I had to turn off "detect thin walls" to get perimeters to show properly so it was an important setting for me. I haven't had the problem above yet though.

xoan, 2015-05-19:
Related: #2829

xoan, 2015-06-17:
Seems to be fixed in 31e0fc7 like the referenced issue.

Thanks.

closed 2015-06-17 by xoan