slic3r/Slic3r #2486: GUI update that can fix the icon padding problem

a4jp-com, 2015-01-02:
Can you make the Windows version of Slicer look like this? 

[image]

Maybe it would be nice to take away the dots after a few words as well.

Add...
Scale...
Settings...
Export STL...
Export G-code... 

This will also fix the padding problem ^^

[image]

xoan, 2015-01-02:
Dots after a word in a button or menu entry means the click opens a dialog or another kind of input to allow user interaction.

koghe, 2015-01-02:
you can customize your Gtk/Wx theme on window too http://gtk-win.sourceforge.net/home/ search for gtk theme on google
dots as to do with some standard good gui design

a4jp-com, 2015-01-03:
Thanks for explaining that Xoan but the save button doesn't have 3 dots so I don't think we need 3 dots to tell us a screen will popup when we hit a button.
If you check out the "save current print settings" icon as an example. It doesn't have the 3 dots.
I can guess when I hit the add icon that a dialogue will pop-up.
Also if I push scale, I just expect a dialogue box of the cursor to change.
Do you think the 3 dots are important? 
I think removing them will make the program cleaner and easier to use.
If they are important I suggest changing the icons instead.
The 3 dots are more confusing and just look like a mistake to me.

a4jp-com, 2015-01-03:
[image]

[image]

alexrj, 2015-01-03:
Short answer to your original question is: no, I can't make the Windows version look like the Linux/GTK one. That depends entirely on the GUI toolkit.

Regarding dots, as @xoan explained they have a standard meaning and their presence is correct. The "Save current print settings" button is just an icon with no text, so putting three dots alone would look pretty dumb.

closed 2015-01-03 by alexrj

a4jp-com, 2015-01-03:
Okay. I understand now. 

If this is a standard rule in programming does Google Chrome only use 3 dots for options with popups?

[image]