slic3r/Slic3r #2488: 1.2.4 current - no brim inside object

koghe, 2015-01-02:
Brim is not generated with object and large empty regions inside

[image]

MorganSandercock, 2015-01-04:
Why not just turn on "raft" for 1 layer? Or modify the original model to include brim where you think it is necessary.

koghe, 2015-01-04:
Hi raft double the print time and is not the right way IMHO to solve the problem.
I can design my own brim, but it take time if you have more objects and/or a complex internal geometry, become a real time consuming.
Maybe is not so simple (in term of code and math) apply the same algorithm for the empyt region inside the object. Not a complete brim but a sort of "sewing stitches" on the perimeter can be enough to avoid the adesion problem.

alexrj, 2015-01-04:
It is intentionally not generated for internal holes. It would be very difficult to remove.

alexrj, 2015-01-04:
Anyway, this is a duplicate of #2026

closed 2015-01-04 by alexrj