slic3r/Slic3r #2492: Feature request - or doc update : How to set default config?

jonandel, 2015-01-03:
several times Ive loaded Slic3r (Windows) and created G-code for my printer, and forgotten to 'load config' - this makes my printer crash the head (unless I happen to be sitting watching it - which isnt a given).  
Can I define the 'default' settings to be my config ?
Could this be a setting in preferences (the default config).

alexrj, 2015-01-04:
Default configuration can't be changed by definition - but I suspect you're misunderstanding how configuration works in Slic3r. No need to do "Load config" at all.

Have you read this?
http://manual.slic3r.org/configuration-organization/configuration-organization

jonandel, 2015-01-04:
Thanks for the reply. Yes, I have profiles saved, so I need to set it so one of these is always used.   I have to remember to load (one of) them, or the g-files aren't right, and it breaks my printer (when I forget, which is easy to do).
Is there a way to use the last profile on next restart of slic3r.?
As mentioned in the post, I can use the load option when invoking the app, but it thinks it's changed for some reason..which doesn't happen when I use 'load config'...
Is there something else I'm not understanding ?

Sent from my Raspberry pi !

alexrj, 2015-01-05:
Slic3r *does* remember the last used profiles when you open it.

I don't understand your last sentence. Can you please provide as many details as you can? Tell us whether you're using Simple Mode or Expert Mode, and the **exact** steps to reproduce your problems. Screenshots are welcome.

jonandel, 2015-01-06:
To demonstrate Slic3r 'losing the config on startup :
1/ Slic3r is located on a server share.  Windows (7 64 bit) have access to this share, and have desktop icons to Slic3r on the share.  The desktop link points to "\\server\3d printer\slic3r\slic3r.exe"
2/ Invoke Slic3r from the shortcut icon.  Slic3r is set to Expert mode.
3/ invoke 'File/load config' - and select the profile  - in my case I want to 'default' to 'Kossel_draft.ini' - (which resides in the same share as the slic3r exe).  Note that Slic3r shows all three configs as 'Kossel_draft'. (see screen shot 1).
4/ import an STL file, and generate the g-code file.  Save this file (anywhere).
5/ close down Slic3r, and follow steps 2 and 4, but miss out step 3.  Note Slic3r has all configs set to '-default-'. (see screen shot 2).
6/ compare the two g-files and they're different, and the '-default-' profile generated g-code causes the head to attempt to exit the printer.

screen shot 1:
[image]

screen shot 2:
[image]

My comments about load option giving unexpected results ...
1/ Update slic3r shortcut to employ the load option pointing to the profile I want to load on start up, by changing shortcut to be : "\\server\3dprinter\Slic3r\slic3r.exe  --load \\server\3dprinter\Slic3r\Kossel_draft.ini"
2/ and on invoking the shortcut, Slic3r thinks the profile has been changed - the "(modified)" in the config sections......
see screen shot 3

[image]

closed 2015-01-06 by jonandel