slic3r/Slic3r #2494: Detect Thin Walls Causes Model Errors

MorganSandercock, 2015-01-04:
My first attempt at printing a spiral-vase object is this: http://www.thingiverse.com/thing:458903 (note: rotate it around the X-axis by 270 degrees to put it the right way up in Slic3r.)  I'm using Slic3r 1.2.4 (I have earlier versions but none of them have the object menu that allows the rocket to be turned upright.)

Playing around with the options, I turned on "Detect Thin Walls" and got this:
[image]
It seems to have *added* extra perimeters, which aren't going to let the spiral work. I did not attempt to print this.

However, **and this is my real problem**, switching the option off seems to somehow damage the model and the smooth circular sections are no longer smooth. This only affects the bottom 20 layers and about 40 layers near the top. These defects are obvious in the actual printed object.
[image]

Playing with it some more, it seems that it may be a poor STL file. Slic3r reports auto-repairing 19 errors. *But why does it get it right the first time?* Switching the option on and off produces different errors each time. Here's what I got with Slic3r's default infill settings after I turned "thin walls" on and off a few times:
[image]

koghe, 2015-01-04:
Hi the model has a lot of errors repairing with blender solve some problems, but is not a good model 
is not modeled to print with the vase option, not all objects can be printed as a vase
a clipping view of the model show the problems
[image]

[image]

MorganSandercock, 2015-01-04:
So why does Slic3r get it right the first time and then each time you change the option it creates different defects?

Vicious-one, 2015-01-05:
This is a long-standing issue, Slic3r messes up the mesh a little more each time the model is re-sliced after changing options. I have to delete and re-add and reposition objects after each option change to get good results.

alexrj, 2015-01-05:
It's the first time I see this reported.

Vicious-one, 2015-01-05:
[image]
[image]
[image]

a4jp-com, 2015-01-05:
I'll try the same file. Do you have the config file?

koghe, 2015-01-05:
Never seen this problem, with a correct model rough is almost gone, maybe you can share config and model?
Here is the rocket remodeled, with no errors
 https://dl.dropboxusercontent.com/u/17373246/rocket-rocket.stl

[image]
[image]

Vicious-one, 2015-01-05:
Here's a simple model, config and screenshots at 0.5 and 3 mm.

http://vserv.sinp.msu.ru/temp/reslice-fail.tar.gz

To reproduce, slice the model, then change some option (for example, vertical shell number) back and forth so Slic3r GUI forgets the toolpaths, reslice, repeat, enjoy.

alexrj, 2015-01-05:
@MorganSandercock and @Vicious-one, does the issue appear if you only change options but never click on "Export G-code…"? Or does it only appear if you export G-code between configuration changes?

alexrj, 2015-01-05:
Okay, I can also reproduce it without exporting G-code.