slic3r/Slic3r #2496: Feature request: please allow to rotate the plater

zapta, 2015-01-04:
I am using Prusa i3V. The x,y origin on the machine is on the top/right corner (when viewing from front) but slc3r's plater view has it rotated by 180 degrees.

Having the display orientation matching the reality of my printer will make it easier to determine what portions of the plater I need to cover with Elmer glue and to optimize the piece orientation for the direction of my material cooling fan (which is on the right).

Also, it would be nice to show in the plater view an indication of the origin and x/y axises.

alexrj, 2015-01-05:
Just rotate your printer (or, better, fix it - you build it the wrong way). In any Cartesian system, origin is in bottom/left.

Origin and axes are displayed in the 3D view (since 1.2.0).

closed 2015-01-05 by alexrj

zapta, 2015-01-05:
>> Just rotate your printer

Rotating will block access to the bed.

>> you build it the wrong way

This is how the Makerfarm Prusa i3v was designed.

>> Cartesian system, origin is in bottom/left.

.. and then somebody invented the rotation matrix, as used in your 3D view.
;-)

Anyway, thanks for considering it.

alexrj, 2015-01-05:
I don't believe that Makerfarm guys did that. Prove it ;-)

whosawhatsis, 2015-01-05:
@alexrj If you check MakerFarm's assembly docs, they do tell you to mount the endstops in the MAX positions (read the procedure on the last page for testing the endstops here: https://drive.google.com/file/d/0B80A_woXoRWdWjZiNHZWOXN1UVU/view), so that they are active when the extruder is to the right and the bed is forward. I'm not terribly surprised that they would do this without configuring the firmware to home in the MAX direction, but it is wrong.

zapta, 2015-01-05:
Look at the firs few minutes here to see where is the home position is
https://www.youtube.com/watch?v=JVA8VjegmVE&feature=youtu.be

Also, at 3:24 the display shows zero for x,y,z at the home position.

QED ;-)

alexrj, 2015-01-06:
Heh.  Well, the instructions are unclear.   But okay, the video shows zero at the home position, so you proved it.  Anyway, being one of the (several) developers of the original Prusa i3, and having built tens myself :-) I can tell you that rotating the coordinate system that way is bizarre and nonsense.

You can fix it quickly. You just need to change a few values in the firmware configuration and reupload it, no need to move endstops:
1. Set X_HOME_DIR to 1 (instead of -1)
2. Set Y_HOME_DIR to the opposite of what it's now (1 if -1 and viceversa)
3. Set INVERT_X_DIR to the opposite of what it's now (false if true and viceversa)
4. Set INVERT_Y_DIR to the opposite of what it's now (false if true and viceversa)

That should be everything you need for fixing your printer and be like everyone else on this planet ;-)

alexrj, 2015-01-06:
Oh, you also need to connect the X and Y endstops to the X-MAX and Y-MAX pins on the RAMPS board (instead of X-MIN and Y-MIN). Basically, just move them by one position towards the Z pins.

zapta, 2015-01-06:
Thanks Alessandro. And also swap the plastic spacer and the support spring
between the old and new home corners of the bed.

>> That should be everything you need for fixing your printer and be like
everyone else on this planet ;-)

Except for all the other Prusa i3V users. ;)  The point is that having the
home point at the rear is handy because it makes the entire bed accessible
to me for example to clean and put Elmer glue.

Given the popularity of the Makerfarm Prusa I3v, please reconsider adding
this configuration flag.

Anyway, thanks for Slic3r, I love it.

Tal

alexrj, 2015-01-06:
>The point is that having the
home point at the rear is handy because it makes the entire bed accessible
to me for example to clean and put Elmer glue.

You're missing one thing: home is **not** necessarily in the same place as origin. With the instructions above, your printer will still home at the rear. But it will have the 0,0 point in the bottom left corner. It's a quite common setup.

zapta, 2015-01-06:
I see your point. Thanks.

It will still be nice to support rotated origins since such machines do
exist in the market.

Tal