slic3r/Slic3r #2497: Overwriting a gcode file causes weird printing results

gabosgab, 2015-01-04:
I'm printing a regular knob and noticed the print only works correctly if I don't overwrite and existing file on the SD card I'm printing from.  Overwriting an existing gcode file causes strange results and the printer runs off to the side. It's almost as if the file is not getting truncated before the file is exported.

Running on Max OSX Yosemite with Slic3r 0.9.9.  Wanted to report in case the file open to export was still set to not truncate the file on export.

alexrj, 2015-01-05:
Have you read the guidelines? This is a very incomplete report.

alexrj, 2015-01-05:
Also, 0.9.9 is a very old version and please, get 1.2.4 and check whether the issue is still there.

gabosgab, 2015-01-05:
Sorry, I haven't read the guidelines. I'll find them and re-open if I can repro this in 1.2.4.

closed 2015-01-05 by gabosgab