slic3r/Slic3r #2498: 1.2.4 Unnecessary retractions during solid infill

mkurc1234, 2015-01-04:
There are some unnecessary retractions triggered during the solid infill. It happens only when a perimeter edge has a certain angle. The infill line is broken and a retraction is triggered. See the screenshot:

[image]

Source model and Gcode are here: https://www.dropbox.com/sh/rvykjdwhy3jf1u9/AAC8l3ndPOBtv9hNtf3K8cTYa?dl=0

I guess that it happens in version 1.2.0 and above. In version 1.1.7 no retraction was triggered, only a travel move was issued. 
I don't know is that a bug or a feature. I think that this behavior only increases the print time and may lead to premature wear of the extruder. 
By the way, why a solid infill can't be done as a single continuous line without any breaks? It is possible for any shape without holes inside.

curiouspl2, 2015-01-04:
noticed this aswell. 
ad breaks - below certain path length , moving extruder back and forth increases risk of picking up recently extruded (hot) extrudate. 
so this makes sense by distributing heat more evenly. 
there are no advanced settings for this feature though , i.e. minimum line length... 
for some materials, like PLA such pattern just increases print time and amount of vibration and noise. 
plus each start-stop usually gets a tiny blob.

alexrj, 2015-01-05:
@mkurc1234, I'll check your files.
Breaking extrusion in those cases is correct: if we connected those lines, we'd have a very sharp angle with a lot of material overlap, thus overstuffing. So this is intentional, and it's even a feature. Retraction shouldn't still be triggered, though.

alexrj, 2015-01-06:
Okay, I fixed this and also improved handling of several other cases.

closed 2015-01-06 by alexrj