slic3r/Slic3r #2500: [feature request] Make Overlap factor, Bridge spacing, etc. tunable

Vicious-one, 2015-01-05:
This should enable to get closer to perfection (proper single wall width and smooth infill at the same time) with various materials, like plywood, trimmer line nylon, no-name ABS with abnormal viscosity.

Maybe this can also help to reduce overextrusion problems and nozzle drag near perimeters when solid rectilinear infill is used in fast accelerated prints.

alexrj, 2015-01-05:
I don't like starting from solutions. I always start from problems. I analyze them, I discuss them, I find potential causes and then finally make hypothesis about possible solutions.
Can we discuss on a concrete test case? With pictures, files, configuration files, tests.

Vicious-one, 2015-01-05:
Working on that...

curiouspl2, 2015-01-07:
i +1 on that one. 
i think 'advanced' settings are all about this, and it allows fine-tuning prints, experimenting, testing... 

perhaps warnings could be put on settings which are likely to disappear with rebuild of flow logic code, but this still helps a lot people who just want to experiment and try to nail problems like badly anchored infills, overfilled solid infill (while rest of print is fine) etc. 

groupthink and brainstorming is all about finding solutions together :)

lordofhyphens, 2016-07-12:
@Vicious-one Did you every get that written up? I see this wasn't updated since.

Vicious-one, 2016-07-12:
@lordofhyphens Overlap factor is tunable now, and the bridge flow formula has been changed, so the described issue is mostly fixed. I think this ticket can be closed for now.

closed 2016-07-12 by lordofhyphens