slic3r/Slic3r #2502: The bumpy circles don't exist in version 1.2.0 but they do in version 1.2.4-dev

a4jp-com, 2015-01-06:
I loaded the same file into the newest version of Slic3r and an older version (1.2.0). In the older version all the circles were perfect. Can you put the code back the way it was?

[image]

http://a4jp.com/3d-files/bugs/2015-01-06th-github-test-clip0001.stl
http://a4jp.com/3d-files/bugs/2015-01-06th-09-05-pm-for-github-config.ini

alexrj, 2015-01-06:
I'd rather fix problems.
Anyway, this a duplicate of #2416...

closed 2015-01-06 by alexrj

a4jp-com, 2015-01-06:
Okay. Sorry I thought my point here was different. I will type in the other thread.