slic3r/Slic3r #2503: Slic3r fails to build latest git

Laura3, 2015-01-06:
Slic3r fails to build at the latest dev: f0de57cbe41489836e194a82ce5e1bea5482fc96

Perl v5.18.2
OS: Ubuntu, 14.04 64bit

````

App::cpanminus is up to date. (1.7023)
Class::XSAccessor is up to date. (1.19)
Encode::Locale is up to date. (1.03)
ExtUtils::MakeMaker is up to date. (7.04)
ExtUtils::ParseXS is up to date. (3.24)
skipping R/RJ/RJBS/perl-5.20.0.tar.gz
File::Spec is up to date. (3.47)
Getopt::Long is up to date. (2.42)
IO::Scalar is up to date. (2.110)
LWP::UserAgent is up to date. (6.06)
Math::PlanePath is up to date. (117)
Module::Build::WithXSpp is up to date. (0.14)
Moo is up to date. (1.006001)
Net::Bonjour is up to date. (0.96)
POSIX is up to date. (1.32)
Scalar::Util is up to date. (1.41)
Test::Harness is up to date. (3.34)
Test::More is up to date. (1.001014)
Thread::Semaphore is up to date. (2.12)
Time::HiRes is up to date. (1.9726)
XML::SAX::ExpatXS is up to date. (1.33)
threads is up to date. (1.96)
Cleaning up build files
Cleaning up configuration files
cpanm (App::cpanminus) 1.7023 on perl 5.018002 built for x86_64-linux-gnu-thread-multi
Work directory is /home/user/.cpanm/work/1420570910.14359
You have make /usr/bin/make
You have LWP 6.08
You have /bin/tar: tar (GNU tar) 1.27.1
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by John Gilmore and Jay Fenlason.
You have /usr/bin/unzip
--> Working on ./xs
Entering /home/user/Scripts/Slic3r-dev/xs
Checking if you have Module::Build 0.36 ... Yes (0.4210)
Running Build.PL
Configuring /home/user/Scripts/Slic3r-dev/xs ... Created MYMETA.yml and MYMETA.json
Creating new 'Build' script for 'Slic3r-XS' version '0.01'
OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have Test::More 0 ... Yes (1.001014)
Checking if you have ExtUtils::Typemap 1.00 ... Yes (1.00)
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.24)
Checking if you have Module::Build 0.3601 ... Yes (0.4210)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Building and testing Slic3r-XS-0.01 ... Building Slic3r-XS
Processing XS typemap files...
Generating main XS file...
cc -I/usr/lib/perl/5.18/CORE -fPIC -xc++ -D_FILE_OFFSET_BITS=64 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/poly2tri/sweep/advancing_front.o src/poly2tri/sweep/advancing_front.cc
cc -I/usr/lib/perl/5.18/CORE -fPIC -xc++ -D_FILE_OFFSET_BITS=64 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/libslic3r/ExPolygonCollection.o src/libslic3r/ExPolygonCollection.cpp
cc -I/usr/lib/perl/5.18/CORE -fPIC -xc++ -D_FILE_OFFSET_BITS=64 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/libslic3r/PlaceholderParser.o src/libslic3r/PlaceholderParser.cpp
cc -I/usr/lib/perl/5.18/CORE -fPIC -xc++ -D_FILE_OFFSET_BITS=64 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/admesh/stl_io.o src/admesh/stl_io.c
cc -I/usr/lib/perl/5.18/CORE -fPIC -xc++ -D_FILE_OFFSET_BITS=64 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/libslic3r/PrintObject.o src/libslic3r/PrintObject.cpp
cc -I/usr/lib/perl/5.18/CORE -fPIC -xc++ -D_FILE_OFFSET_BITS=64 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/libslic3r/Extruder.o src/libslic3r/Extruder.cpp
cc -I/usr/lib/perl/5.18/CORE -fPIC -xc++ -D_FILE_OFFSET_BITS=64 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/clipper.o src/clipper.cpp
cc -I/usr/lib/perl/5.18/CORE -fPIC -xc++ -D_FILE_OFFSET_BITS=64 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/libslic3r/Layer.o src/libslic3r/Layer.cpp
cc -I/usr/lib/perl/5.18/CORE -fPIC -xc++ -D_FILE_OFFSET_BITS=64 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/admesh/stlinit.o src/admesh/stlinit.c
[... 117 lines cut ...]
Failed 3/20 subtests 
t/svg.t ........................ ok   
t/thin.t ....................... ok     
t/threads.t .................... ok   
t/vibrationlimit.t ............. ok   

Test Summary Report
-------------------
t/combineinfill.t            (Wstat: 6400 Tests: 3 Failed: 0)
  Non-zero exit status: 25
  Parse errors: Bad plan.  You planned 8 tests but ran 3.
t/retraction.t               (Wstat: 256 Tests: 19 Failed: 1)
  Failed test:  19
  Non-zero exit status: 1
t/skirt_brim.t               (Wstat: 6400 Tests: 3 Failed: 0)
  Non-zero exit status: 25
  Parse errors: Bad plan.  You planned 5 tests but ran 3.
t/support.t                  (Wstat: 6400 Tests: 17 Failed: 0)
  Non-zero exit status: 25
  Parse errors: Bad plan.  You planned 20 tests but ran 17.
Files=33, Tests=374, 116 wallclock secs ( 0.22 usr  0.05 sys + 127.14 cusr  0.83 csys = 128.24 CPU)
Result: FAIL
Some tests failed. Please report the failure to the author!

````

lordofhyphens, 2015-01-06:
known failure, see https://travis-ci.org/alexrj/Slic3r

Laura3, 2015-01-06:
So there's already automated build checking. So I shouldn't bother reporting like this next time?

alexrj, 2015-01-06:
Thank you @Laura3!

Well, Travis CI notifies build/test errors to me. I usually fix them quickly.
If you notice that I don't fix them for too long time, or that Travis CI reports success while you have local failures, your reports are more than welcome!

closed 2015-01-06 by alexrj