slic3r/Slic3r #2505: Add Machinekit to G-code flavors

strahlex, 2015-01-07:
Currently Slic3r has the option "Mach3/LinuxCNC" for GCode flavor. As Machinekit is fork of LinuxCNC for embedded platform and mainly used for 3D printing it makes sense to add it to the G-Code flavors. 

The output should be the same as for "Mach3/LinuxCNC", particularly the E axis needs to be renamed to A.

strahlex, 2015-01-07:
Link to the project website: http://www.machinekit.io/

strahlex, 2015-01-07:
The Mach3/LinuxCNC flavor option uses the P parameter for M1xx codes, however. Machinekit supports the RepRap style S parameter.

Can someone please point me to where the GCode flavors are generated?

alexrj, 2015-01-28:
@strahlex, it looks like some more codes need to be changed for Machinekit:
https://github.com/strahlex/Machinekit-Slic3r-postprocess/blob/master/postprocess.py

If so, can you update your pull request with the missing bits? Thank you! :)

strahlex, 2015-02-06:
Ok, I added the retract and unretract commands.

Is there a way to automatically add a M2 command at the end of the GCode file? The M2 is necessary for all Machinekit flavour GCode files.

strahlex, 2015-02-07:
Ok I found it. I added a flavour specific funtion to the GCodeWriter that automatically adds the M2 code at the end of the program.

alexrj, 2015-02-15:
Merged. Thank you!

closed 2015-02-15 by alexrj