slic3r/Slic3r #2506: Adjacent bridge areas not being combined

merkur2k, 2015-01-07:
I have a case where I have two bridge areas close enough to each other that they overlap. Instead of combining it into one bridge, it is doing them one at a time which leaves the edge of one over open air causing it to fail.
Slic3r 1.2.4 Windows 7 x64.
Here is the layer below the bridge:
[image]
And the bridge layer:
[image]

mkurc1234, 2015-01-10:
I have noticed this issue is present still in 1.2.5. I had tried to print a T shaped object with my own support structure and this is what i got:

Layer before bridging:
[image]

Bridging layer:
[image]
The bridge structure should span the whole layer surface while it changes abruptly not even on perimeter boundary but in mid air.

Layer above bridges:
[image]

BR,
Maciej

harriv, 2015-01-10:
Please add information about your settings and STL files, otherwise it's very hard to find out what's wrong. See the guidelines: https://github.com/alexrj/Slic3r/blob/master/CONTRIBUTING.md

mkurc1234, 2015-01-10:
Here it is: https://www.dropbox.com/sh/9je1p1o7ku1qao4/AACMZI6zYRfJ51cV6oKWF1ADa?dl=0

lordofhyphens, 2017-03-19:
The testcase is a dead link, and I believe this has been fixed in the intervening time. 

Will reopen if a test case appears (please upload here according to the directions at https://help.github.com/articles/file-attachments-on-issues-and-pull-requests/)

closed 2017-03-19 by lordofhyphens