slic3r/Slic3r #2507: Filament diameter not read-in

docMP, 2015-01-07:
I have used parallel Slic3r 117 and 124. The later fails to print thin walls, like 2x1x6 lego brick (http://www.thingiverse.com/thing:5699). The same OpenSCAD generad code procuces with same settings in Slic3d, where main difference is that the material consumtion in the v124 is 119mm and v117 is 350 mm. Major difference in codes:
v124:
; filament_diameter = 3,3,3 (3 toos set)

v117:
; filament_diameter = 1.75 (1 tool set, multiple tools do not work)

Using RepRapPro TriColor Medel, 1.75 mm filament, 0.5 mm nozzle

alexrj, 2015-01-08:
Wait.  Let's talk about 1.2.4.  I don't support 1.1.7 anymore.

Can you please provide all the information needed? It's clearly written in the guidelines:
https://github.com/alexrj/Slic3r/blob/master/CONTRIBUTING.md

docMP, 2015-01-10:
Dear Alessandro,Thanks from quick reply, sorry for my delay. I did not manage attc. files to my report, so please find them attc here. If I should attc them to the discussion, any instructions would be appreciated in this UI.About profiles: I have gone trough the instructions in reprapro site, and downloaded their .ini file, butbut.. there seems still be some version differences, as I'm not sure where to save, which ini, and which selection should save all(?) information. Anyway, one ini-file is attc too.In program I have 3 heard, 0.5 mm nozzle, 1.75 mm filement (multiplier 1). Almost all paremeters looks to be similar. Some starting layers details are different here.If this information is not sufficient, please let me know in detail what is needed.
MP

   

https://github.com/alexrj/Slic3r/blob/master/CONTRIBUTING.md—

a4jp-com, 2015-01-10:
Maybe it's better to remove the email address above so you don't get spam.

alexrj, 2015-01-16:
@docMP, e-mail attachments don't work with GitHub. You need to host your files somewhere (Dropbox, wetransfer, whatever) and put a link here.

docMP, 2015-01-31:
Hi,
Sorry long delay here. Now I have v1.2.5, but the same problem continues with multiple heads. Examples of similar parameters with 117 (tool 1 used(T0)) and v125  (tool 2 used (T1)) 
at http://we.tl/b7lRnsMS7N
where later with 3 filaments.
; filament_diameter = 1.75,3,3
... yes, it assumes 3 mm, and there is no place to enter the other diameters.

v125 ; filament used = 64.5mm (0.5cm3) (T1 used)
v117 ; filament used = 208.2mm (0.5cm3) (1 tool use only)
and if I move the work back to 1 filament/tool at v125
; filament used = 185.3mm (0.4cm3) = better.... at  http://we.tl/bYoAN9v7As

Further the code heats with V125 the tool T0, even if it is not used.

docMP, 2015-02-22:
The problem continued in v1.2.6, when using multiple head tool (TriColorMendel on Ubuntu): 
If two Extruders are selected, the "cube([10,10,10]);" on two locations with two different extruder selected for them (via Plate/Settings/Parts/Object..cube.stl->Extruder "1" and other "2"), resulted code shows:
; filament used = 205.8mm (0.5cm3)
; filament used = 65.3mm (0.5cm3) 
..
; filament_diameter = 1.75,3    

the 3 above is not anywhere in my settings.
Maybe the "Filament Setting"/Filament/Diameter should be moved to under individual Extruders, as it is possible to build system with different width filaments.

Furthermore, in the same setting box: "Extrusion Multiplier" change to "2", in order to go around the previous issue, has no effect to the code generated.

markwal, 2015-02-22:
Apologies if I'm misunderstanding, but I have a dual extruder and am able to choose a filament preset for each extruder and it comes through as:

; filament_diameter = 1.72,1.72

in the gcode output.  

I'm using the GUI in expert mode and I get to select the filament preset for each extruder on the right and side of the Plater window.

alexrj, 2017-03-17:
I'll close this as it's very old and a LOT of code was written since then. Multiple extruder support is now mature and known to work well. If you need further assistance feel free to reopen this, we'll do our best.

closed 2017-03-17 by alexrj