slic3r/Slic3r #2512: 1.2.4-dev: Plater is saving STLs in the air again

lordofhyphens, 2015-01-08:
Original files: 
http://www.thingiverse.com/download:241326
http://www.thingiverse.com/thing:262121

Added two copies of the second item and one of the first, rotated the first to make it flat on the bed, and then saved as an STL. Added the exported stl and the two ducts were hanging in the air.

Current workaround is to split the STL, which puts everything on the plate properly again.

Resulting STL:
STL: http://www.2shared.com/file/SHlNrmUd/fan.html

lordofhyphens, 2015-01-08:
Remember the second download button is the correct one, first (larger) one is an ad.

alexrj, 2015-01-08:
:(

lordofhyphens, 2015-01-08:
Not trying to make life harder for you :)
I'm in #slic3r if you need to chat directly about it.

alexrj, 2015-01-08:
Heh, I really appreciate your reports. Sad face was because I was hoping this issues about floating models were fixed.

lordofhyphens, 2015-01-08:
If it helps, the models aren't floating in the gcode export.

alexrj, 2015-01-08:
Fixed!

closed 2015-01-08 by alexrj

lordofhyphens, 2015-01-09:
Time to look for more bugs then :D

alexrj, 2015-01-09:
Heh :P