slic3r/Slic3r #2513: Overlapping gap fill

Vicious-one, 2015-01-09:
Sometimes Slic3r generates a self-overlapping gap fill. 
The problem may disappear when object orientation is changed or slicing resolution is manually limited.

[image]

Example model with bugged top slice, config, STL and G-code are here:
http://vserv.sinp.msu.ru/temp/gap-overlap.tar.gz

alexrj, 2015-05-04:
@Vicious-one, I'm actually checking your G-code file but I can't find any overlap in the uppermost layer… can you help me find where it is?

Vicious-one, 2015-05-04:
Line 9900:

G1 E3510.09983 F1200.00000
G1 X-8.593 Y30.121 E3510.77640 F1800.000
G1 X23.045 Y30.102 E3511.44790

alexrj, 2015-05-22:
This should be fixed by #2836

Vicious-one, 2015-05-22:
Seems fixed, but I'll test this on some more hardcore manifestations, of which I already have plenty.

alexrj, 2015-05-22:
Okay, thank you. Feel free to report them!

closed 2015-05-22 by alexrj