slic3r/Slic3r #2515: Incorrect slicing, leaving gaps

dexy31337, 2015-01-09:
Greetings, I've found strange Slic3r behavior (tried both 1.1.7 and 1.2.4) while slicing model http://www.thingiverse.com/thing:13188. The problem in that instead of making solid walls which should be 2-line thick, on impeller blades, sometimes there appear gaps and one-line areas. I've sliced and printed same model in Simplify3D, and there are no such problems.
I've made a cut in blender at problematic  place in blender (@21.3mm, layer 142 in g-code, with layer height 0.15mm) and noticed that there are indeed edges where Slic3r tries to make empty spots.
Please see [attachments] ( https://www.dropbox.com/sh/2lb5pqgz5j2u3fc/AABQmsPIkcPdZimqCIhkZJSNa?dl=0)

walruses, 2015-01-10:
Have you tried disabling "detect thin walls"?

dexy31337, 2015-01-10:
@walruses thanks! Works like a charm. My bad, I didn't notice this option...

closed 2015-01-10 by dexy31337

bstott, 2015-01-25:
As you read from my recent post in #2560 I have the problem created by making a thin layer height (0.24mm). I have turned off the 'detect thin walls' and can create a filled wall with setting the layer height above 0.32mm. I do not know the transition thickness which creates this issue.

GHPS, 2015-01-25:
Indeed whether or not "detect thin walls" works correctly depends on the layer height and and the number of perimeters.

Using the same object as above and detection turn on I get a gap on the top layer (4.00) when either
a) the perimeter is greater than 3 (with layer height 0.1) or
b) a layer height less than 0.25 (with a perimeter number 3)