slic3r/Slic3r #2518: Chrashing importing STL files in 1.2.5

TrevorTurner, 2015-01-10:
I am experiencing slic3r crashing when importing STL files generated in FormZ. The application closes after selecting the file with 'Add'. Please provide me with a way to send you and example file that crashes on adding. I am using Windows 7 64 Bit

dexy31337, 2015-01-10:
@TrevorTurner (dropbox)[dropbox.com] is a good way of sharing files.

dexy31337, 2015-01-10:
@TrevorTurner [Dropbox](http://www.dropbox.com) is a good way of sharing files.

TrevorTurner, 2015-01-10:
It appears dropbox can only share with specific email addresses

TrevorTurner, 2015-01-10:
An example of a file causing crashing on adding is:
https://www.dropbox.com/s/05m3hdrmrw193qp/strap1_6.stl?dl=0

braddo99, 2015-01-10:
I also have a file which crashes 1.2.5 on Windows 7 64 on opening an STL file. Here is a link to the STL: https://www.dropbox.com/s/kcnxbvmq4kxnt5r/Brad_fixed.stl?dl=0

Here is the config file: 
https://www.dropbox.com/s/8nqmkd9yyqo6ncp/config_with_crash_on_open.ini?dl=0

lordofhyphens, 2015-01-13:
@TrevorTurner Does it crash on 1.2.4 or 1.2.3? It's a lot fewer commits to walk through if it works with either of those two versions.

dexy31337, 2015-01-14:
Just for info - I've opened both STL files on Mac and Win7x32 with no crash with 1.2.5.

alexrj, 2015-01-18:
I think this might be related to #2517, so it needs to be tested with the upcoming 1.2.6.

TrevorTurner, 2015-01-18:
Sorry did not get a chance yet to test on earlier versions yet. Will also test on 1,2.6 when it is available.

alexrj, 2015-02-01:
Hello, can you confirm this is fixed in 1.2.6? Thank you!

TrevorTurner, 2015-02-01:
Still crashing immediately on add in 1.2.6 on 64 Bit Windows 7

TrevorTurner, 2015-02-01:
Also crashes on 1.2.3 64 Bit Windows 7

alexrj, 2015-03-09:
I'm not able to reproduce this with 1.2.6 or 1.2.7-dev... :(

braddo99, 2015-03-22:
I am still experiencing crashes with 126 but I think it is due to importing setting files from previous versions which are somehow incompatible. For example, I did a new install of 126, imported the settings file I linked above, then imported the linked stl file. Crash. However, if I delete the C:\Users\Me\Roaming\Slic3r folder, re-open without the settings (redo manually) then the file opens, exports and prints just fine.

alexrj, 2015-06-01:
@TrevorTurner, @braddo99: this should be fixed with 1.2.7 finally.  Can you confirm so that we can close this report? Thank you in advance!

braddo99, 2015-06-06:
OK, let me try to recreate the above...

alexrj, 2015-06-20:
@braddo99, let me know if 1.2.9 fixes the issue and we can close this one. Thank you!

alexrj, 2016-03-18:
@braddo99, no answer from you.  I assume this is fixed.

closed 2016-03-18 by alexrj