slic3r/Slic3r #2519: [Feature request] Optimize temperature settings

dexy31337, 2015-01-10:
Hi.
I have two requests to consider.

First. I was printing PLA with 60° for first layer, and turning it off for others, so my options were:
`bed_temperature = 0`
`first_layer_bed_temperature = 60`
It took me some time to realize (and check manual to verify), that setting `bed_temperature = 0` actually doesn't generate g-code as I expected (I was waiting for M109 S0). Now, I'm using workaround, setting `bed_temperature = 1` to cool down HBP. This is quite confusing. Maybe it's good idea to add checkbox so, user could check it if he doesn't want temperature management at all, and handle 0 value as a temperature control command?

Second. Why limit setting temperature to only first and other layers? Why not allow user to enter multiple values, to allow any temperature to be set at any layer? This would enhance flexibility and give more control over printing.

lordofhyphens, 2015-01-13:
Assuming unsigned ints aren't used, might be better to reserve -1 for "don't do anything". That being said, any number below your ambient temperature is going to have the same real effect. 

For the second item, UI clutter is a real thing and a clever post-processing script will give you a lot of the effects you want, parsing the file and inserting G code as necessary -- wouldn't be too difficult to put a GCODE comment that has the interesting values in your start gcode that the script could pick up on and tweak things as you want.

I've also not seen a real benefit to stepping up/down on bed heat outside of the start/end GCODE personally.

dexy31337, 2015-01-13:
@lordofhyphens I agree, that interface should be kept nice and clean, but that is a UI/UX question, and should be considered apart from functionality.

Of course g-code is easy to write manually, or with script, but having option to save everything you need in profile, is always nice and simplifies workflow.

As of practical benefits, i can recall two simple cases:
1. Sometimes I prefer to change bedplate/filament temperature after several layers, instead of after first one.
2. When printing with different layer thickness, it may be useful to change temperature a bit, depending on filament.

lordofhyphens, 2017-03-27:
@dexy31337 I've bee putting some more thought into this, and I think that a modifier mesh attribute to apply a difference to the current temperature would work out decently.