slic3r/Slic3r #2520: 1.2.5 After G11 firmware unretract not resetting E to 0

braddo99, 2015-01-10:
I am using Windows 7 64 and noticed firmware unretract is putting huge blobs of plastic on the bed early and often :-)

Here's a snippet from the beginning of a recent gcode output: Notice right after the unretract, there is an E92 S0 and then immediately E 6.94.. I think this is the problem.

; generated by Slic3r 1.2.5 on 2015-01-09 at 21:10:04

; external perimeters extrusion width = 0.50mm
; perimeters extrusion width = 0.50mm
; infill extrusion width = 0.50mm
; solid infill extrusion width = 0.50mm
; top infill extrusion width = 0.18mm

M107
M190 S110 ; set bed temperature
M104 S248 ; set temperature
G28 ; home all axes
G1 X-70 Y25 Z5 F5000 ; prepare for wipe
G0 Z0.1 F500;
G91
G1 E2
G90
G0 Y0;
G1 Z5 F5000;

M109 S248 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G10 ; retract
G92 E0
G1 Z1.500 F8400.000
G1 X55.770 Y0.000 F8400.000
G1 Z0.300 F8400.000
G11 ; unretract
G92 E0
G1 X55.770 Y60.000 E6.94114 F3000.000
G1 X55.374 Y62.102 E7.18864
G1 X54.239 Y63.916 E7.43613

braddo99, 2015-01-10:
It's interesting that at about line 8000, Z5.3 this issue is no longer present. But the files are unprintable with firmware retract.

wolfmanjm, 2015-01-10:
Aww crap, I thought the blobs were bad tuning :( been trying to get width/ht right and it turns out to be a bug :) Yea I just looked at the last few gcode files and I see similar issues, large extrusions after retract.

G92 E0
G1 Z0.400 F18000.000
G10 ; retract
G92 E0
G1 X0.000 Y66.114 F18000.000
G11 ; unretract
G92 E0
G1 X-59.000 Y66.114 E26.29412 F1800.000 <--- Yikes!
G1 X-61.344 Y65.717 E27.35358

However that is the very first one, after that it settlles down and seems to get it correct for subsequent retracts.

wolfmanjm, 2015-01-10:
Actually I thought the same thing, but I don't think it is a bug, look at the length of this move in your example... from X55.770 Y0.000 to X55.770 Y60.000 that is a long move and 6.6mm of filament does not seem unreasonable in that case. Maybe? Anyway I lloked at my recent gcode and I saw similar things but when I looked at the length of the move it was about the right amount of extrusion.

braddo99, 2015-01-10:
Good point, the object is just a long box - I compared to a 1.7 stable gcode of the same stl and I do see similar large extrusion calls so I guess this is normal. However, there is something else wrong because the blobs (I'm talking solid penny sized) are there. A couple of other things I didn't mention above which are probably important is that I'm using a DELTA and running Smoothie firmware. These files could somehow be injured though because they are all exported just prior to a Slic3r hard crash.

wolfmanjm, 2015-01-10:
yea with smoothie you must safely eject the sdcard after writing to it otherwise the gcode file you copied will be truncated or corrupted. Also with smoothie make sure that you unmount the sdcard from the host before printing as it can cause pauses which will cause blobs.
Lastly check in smoothie what your extruder accleration is, the default used to be 50 which was way too slow and should be at least 500. also double check the firmware retract settings. go to #smoothieware on IRC for more help.

braddo99, 2015-01-11:
Hey Wolfman, thanks for the reminders. All of these things are OK, been using Smoothie very successfully for several months. I'm not writing to the SD, running from Octoprint. This is only an issue with Slic3r 1.2.4 and 1.2.5 - I was eager to try out the advance, but it's not in the cards ATM.

alexrj, 2015-01-13:
@braddo99, when a `G92 E0` is written, the E value is reset internally. Also, in the snippet you posted I can't see any extrusion before the `G11`, so E was 0 anyway (nothing to reset it from).

If you think that 6mm of filament is too much for a 60mm move you might want at least to post your configuration. Read guidelines. An incomplete report is not useful...

alexrj, 2015-01-16:
@braddo99, what's the status of this one?
Can we close it or do you still think there's a bug that needs to be investigated?

braddo99, 2015-01-16:
I just finished printing a big project so can go back to testing 1.2.5. However, I never made a successful print with that version. 1.2.4 had both massive retraction (filament would come out of the extruder) and massive unretract, big blobs. 1.2.5 has just the big blobs. I'll try some more tonight!

alexrj, 2015-01-18:
Unless there are proofs that a bug exists, blobs are most likely caused by bad configuration.
Let me know about this status of this particular issue. Can we close it or not?

braddo99, 2015-01-21:
I still have this issue but I should "retract" what I said earlier, it's not an issue with only firmware retract but with (firmware retract and) pressure advance. I set pressure advance to 5 (don't have any idea where to start with this number)

Here are the first few lines of a 50mm thin wall calibration object gcode as sliced with 1.2.5:

; generated by Slic3r 1.2.5 on 2015-01-21 at 16:15:01

; external perimeters extrusion width = 0.50mm
; perimeters extrusion width = 0.85mm
; infill extrusion width = 1.02mm
; solid infill extrusion width = 0.85mm
; top infill extrusion width = 0.85mm

M107
M190 S100 ; set bed temperature
M104 S248 ; set temperature
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle

M109 S248 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G10 ; retract
G92 E0
G1 Z0.300 F8400.000
G1 X30.536 Y0.000 F8400.000
G11 ; unretract
**G92 E0
G1 E41.82818 F1800.000 ; pressure advance
G92 E0.00000 ; restore E**
G1 X30.536 Y25.000 E4.01714 F1080.000
G1 X30.139 Y27.057 E4.35376
G1 X29.007 Y28.819 E4.69037
G1 X27.301 Y30.035 E5.02699
G1 X25.000 Y30.536 E5.40532
G1 X-25.000 Y30.536 E13.43960
G1 X-27.057 Y30.139 E13.77621
G1 X-28.819 Y29.007 E14.11283
G1 X-30.035 Y27.301 E14.44945
G1 X-30.536 Y25.000 E14.82778
G1 X-30.536 Y-25.000 E22.86206
G1 X-30.139 Y-27.057 E23.19867
G1 X-29.007 Y-28.819 E23.53529
G1 X-27.301 Y-30.035 E23.87190
G1 X-25.000 Y-30.536 E24.25023
G1 X25.000 Y-30.536 E32.28451
G1 X27.057 Y-30.139 E32.62113
G1 X28.819 Y-29.007 E32.95775 F1080.000
G1 X30.035 Y-27.301 E33.29436
G1 X30.536 Y-25.000 E33.67269
G1 X30.536 Y-0.075 E37.67778
G1 E-4.15040 F1800.000 ; pressure discharge
G92 E37.67778 ; restore E
G10 ; retract
G92 E0
G1 X23.629 Y-23.629 F8400.000
G11 ; unretract
G92 E0
G1 E56.93159 F1800.000 ; pressure advance
G92 E0.00000 ; restore E
G1 X23.629 Y23.629 E7.59360 F1260.000
G1 X-23.629 Y23.629 E15.18721
G1 X-23.629 Y-23.629 E22.78081

alexrj, 2015-01-28:
It's correct. Since you use a 1.12838mm filament diameter, you need to use a much lower factor for Pressure Advance.

closed 2015-01-28 by alexrj

braddo99, 2015-01-29:
OK, I wondered if the volumetric extrusion "trick" of using 1.12mm diameter to get extrusion in mm3 would in some way interfere with other aspects like pressure advance. I'll have to experiment further there!

alexrj, 2015-01-29:
You have now an option for Volumetric Mode, so no need for the filament diameter trick

Lenbok, 2015-02-07:
Two questions:

- Does using the option for Volumetric Mode change the interpretation of the Pressure Advance factor?

- Any idea what is an appropriate default for Pressure Advance when using volumetric extrusion (either via the "diameter trick" or the new Volumetric Mode) with a non-bowden extruder?