slic3r/Slic3r #2521: Any model gets chopped off at the bottom of the screen when panning (1.2.5-dev Windows 7 64-bit)

a4jp-com, 2015-01-10:
Any model gets chopped off at the bottom of the screen when panning (1.2.5-dev Windows 7 64-bit)

[image]

a4jp-com, 2015-01-11:
[image]

a4jp-com, 2015-02-08:
The newest version of Slic3r still has a panning problem and models get hidden or cut off when they are moved or rotated. How can I pan or rotate my model without it getting cut up/hidden?

This video below just shows the problem without a model. I'm using Windows 7 64-bit. 

[image]

a4jp-com, 2015-02-18:
Any updates for this bug yet? My screen is 2560 x 1440. Maybe that is a problem...

gege2b, 2015-02-18:
It looks like some sort of camera clipping issue

I don't have this problem on my win7 x64 machines, may be you could try with a lower resolution (my screen is at 1920 x 1080)

a4jp-com, 2015-02-18:
I just set my screen resolution to 1920 x 1080 and still get the same problem.

[image]

a4jp-com, 2015-02-20:
I'm using an Nvidia Quadro 5000 card with the latest driver. What type of video card do you have gege2b? Do you think the problem could be graphics card related?

gege2b, 2015-02-20:
I've a Nvidia too, GeForce GTX560M but not with the latest drivers (They're downloading right now)
I don't know if its related to graphic card and/or drivers, do you see this behavior in other apps or games ?

gege2b, 2015-02-20:
Well I'd badly tested
I got this problem too ! I'd just not zoomed far enough ^_^

So, when the zoom is far enough, this problem occurs
It's like if the canva used to render the 3D view is too small.

May be a (dirty) workaround should be to limit the zoom-

a4jp-com, 2015-02-20:
I've only had this problem in Slic3r and in so many different versions. It would be nice to get this fixed as my models always get cut up when trying to check them.

a4jp-com, 2015-02-20:
Limiting the zoom wouldn't help much but moving the canvas layers the right way around might fix this or changing the start position. If the model was higher up it might be a quick fix until the coding is fixed.

gege2b, 2015-02-21:
if you drag the view to be almost parallel to the plane, it's worth

seems like the bounding box around the plane (+ axis) is clipping the view

a4jp-com, 2015-02-24:
About 30 percent of my window is unusable. :( I've tried re-sizing and moving the window but nothing helps. If I make the window shorter by re-sizing it the area that gets cut off gets smaller but then I only have a small area I can work in. How do I fix this problem? Does everyone else have the same problem as well? I have two 2560 x 1440 monitors on Windows 7 (64-bit). The Nvidia driver for the Quadro 5000 video card is 347.52.

gege2b, 2015-02-24:
I have this problem on linux too (debian jessie), in addition to my two different Win7 x64 (my debian is installed as dual boot on one of the Win7)

So I don't think it's related to the graphic card or drivers

a4jp-com, 2015-02-24:
Maybe I will have to play around with the code to see if I can find where the border of the drawing area is...

a4jp-com, 2015-02-28:
Is there anyone out there that can make a patch for this? I'll keep trying but I haven't come up with anything yet.

lordofhyphens, 2017-03-19:
@a4jp-com is this still a problem? I can't reproduce on 1.3.0-dev.

http://dl.slic3r.org/dev

a4jp-com, 2017-03-20:
Looks like it's fixed in the latest development version. Thank you ^^

gege2b, 2017-03-21:
looks fixed here too thanks !