slic3r/Slic3r #2522: Extruder assignment in multimaterial printing not respected

pentaglut, 2015-01-10:
When I load a multimaterial file, I need to map the different materials of the model to my extruders. There seems to be a nice menu for that in Part settings -> extruder.
However this choice is not respected by slic3r, and extruders are always arbitrarily assigned.
(ver 1.2.5)

alexrj, 2015-01-13:
I just tried, and it appears to work for me.
Can you provide detailed steps for reproducing this?

pentaglut, 2015-01-14:
I built an amf file with slic3r by combining the 3 slt files of the tricolour cone: http://www.thingiverse.com/thing:49702/#files
Then I loaded it into slic3r and assigned a material to each body:
Body 0 = extruder 1
Body 1= Extruder 2
Body 2= EXtruder 3
[image]
I opened the file with repetier. And the results are matching (blue is extruder 1, gold is 2, grey is 3).
[image]

But when I change the assigned materials from the default scheme and give extruder 3 to body 0:
[image]
I see no modification in the resulting gcode file:
[image]

I have the same problem when using repetier to assign materials.
I tested with individual objects (not an amf file) and extruder choice worked well.

Here are my print settings:
; avoid_crossing_perimeters = 0
; bed_shape = 0x0,215x0,215x195,0x195
; bed_temperature = 105
; bridge_acceleration = 0
; bridge_fan_speed = 100
; brim_width = 0
; complete_objects = 0
; cooling = 1
; default_acceleration = 0
; disable_fan_first_layers = 3
; duplicate_distance = 6
; end_gcode = M104 S0\nM140 S0\nM0\nG1 X0 Y180 F12000\nM84
; extruder_clearance_height = 20
; extruder_clearance_radius = 20
; extruder_offset = 0x0,0x0,0x0
; extrusion_axis = E
; extrusion_multiplier = 0.92,0.92,0.92
; fan_always_on = 0
; fan_below_layer_time = 18
; filament_diameter = 1.75,1.75,1.79
; first_layer_acceleration = 0
; first_layer_bed_temperature = 105
; first_layer_extrusion_width = 0
; first_layer_speed = 50%
; first_layer_temperature = 240,240,240
; gcode_arcs = 0
; gcode_comments = 0
; gcode_flavor = reprap
; infill_acceleration = 0
; infill_first = 0
; layer_gcode = 
; max_fan_speed = 100
; min_fan_speed = 30
; min_print_speed = 10
; min_skirt_length = 12
; notes = M-PLA-05-03-1
; nozzle_diameter = 0.32,0.32,0.32
; only_retract_when_crossing_perimeters = 0
; ooze_prevention = 1
; output_filename_format = [input_filename_base]_ABS.g
; perimeter_acceleration = 0
; post_process = 
; pressure_advance = 1
; resolution = 0.02
; retract_before_travel = 2,2,2
; retract_layer_change = 1,1,1
; retract_length = 2.5,2.5,2.5
; retract_length_toolchange = 5,5,5
; retract_lift = 0.2,0.2,0.2
; retract_restart_extra = 0,0,0
; retract_restart_extra_toolchange = 0,0,0
; retract_speed = 80,80,80
; skirt_distance = 3
; skirt_height = 999
; skirts = 1
; slowdown_below_layer_time = 5
; spiral_vase = 0
; standby_temperature_delta = -15
; start_gcode = M0\nM190 S[first_layer_bed_temperature]\nM104 S50 T0 ; set temp pour que slic3r ne le fasse pas\nM104 S50 T1\nM104 S50 T2 \nG10 P0 S[temperature_0] R220\nG10 P1 S[temperature_1] R220\nG10 P2 S[temperature_2] R220\nG28 X0 Y0\nG28 Y0 Z0\nG1 X0 Y20 Z0.3 F500\nG1 X15 Y20 F12000
; temperature = 240,240,240
; threads = 2
; toolchange_gcode = 
; travel_speed = 250
; use_firmware_retraction = 0
; use_relative_e_distances = 0
; use_volumetric_e = 0
; vibration_limit = 0
; wipe = 0,0,0
; z_offset = 0.06
; dont_support_bridges = 1
; extrusion_width = 0
; first_layer_height = 0.2
; infill_only_where_needed = 0
; interface_shells = 0
; layer_height = 0.2
; raft_layers = 0
; seam_position = aligned
; support_material = 0
; support_material_angle = 0
; support_material_enforce_layers = 0
; support_material_extruder = 1
; support_material_extrusion_width = 0
; support_material_interface_extruder = 2
; support_material_interface_layers = 2
; support_material_interface_spacing = 1
; support_material_interface_speed = 100%
; support_material_pattern = rectilinear
; support_material_spacing = 4
; support_material_speed = 20
; support_material_threshold = 45
; xy_size_compensation = 0
; bottom_solid_layers = 3
; bridge_flow_ratio = 1
; bridge_speed = 30
; external_fill_pattern = rectilinear
; external_perimeter_extrusion_width = 0
; external_perimeter_speed = 26
; external_perimeters_first = 0
; extra_perimeters = 1
; fill_angle = 45
; fill_density = 15%
; fill_pattern = honeycomb
; gap_fill_speed = 20
; infill_every_layers = 1
; infill_extruder = 1
; infill_extrusion_width = 0.4
; infill_speed = 50
; overhangs = 1
; perimeter_extruder = 1
; perimeter_extrusion_width = 0
; perimeter_speed = 32
; perimeters = 2
; small_perimeter_speed = 26
; solid_infill_below_area = 5
; solid_infill_every_layers = 0
; solid_infill_extruder = 1
; solid_infill_extrusion_width = 0
; solid_infill_speed = 40
; thin_walls = 1
; top_infill_extrusion_width = 0.32
; top_solid_infill_speed = 32
; top_solid_layers = 4

alexrj, 2015-01-14:
Thank you for the details. However I still can't reproduce it: when I modify the extruder assigned to first part, I get a correct G-code file.
One thing that differs in my test from your screenshots is that parts are not named "0", "1", "2" but they get the original file names. How did you build the AMF file? Maybe you used a pre-1.2.0 version?

pentaglut, 2015-01-15:
Yes, the file is created with ver 1.1.7:  <metadata type="cad">Slic3r 1.1.7</metadata>
I also of this problem with files created by repetier (I think it creates a temporary AMF file that gets opened by slic3r in command line). You can see that in my issue 2523. I must however admit that I haven't checked if the problem comes from repetier.

alexrj, 2015-01-16:
Good to know, thank you. I think that's because AMF files generated by previous Slic3r versions assigned extruders to materials instead of volumes. I'll check. It would be good to ensure backwards compatibility.
Can you try creating the AMF file directly from 1.2.5?

alexrj, 2015-01-16:
Okay, I was able to reproduce this by creating an AMF file with Slic3r 1.1.7.

alexrj, 2015-01-16:
I pushed a change that makes Slic3r compatible with old AMF files. Thank you for reporting this.

closed 2015-01-16 by alexrj