slic3r/Slic3r #2524: Sequential Printing - Collision when using "Increase Copies"

jeremyferris, 2015-01-11:
When trying to create multiple objects in Slic3r 1.2.5 using the "Increase Copies" (Ctrl+ +) and then performing a sequential print, the generated G-Code includes one move into a previously printed object (presumably the original copy) just after moving Z down to the base layer. This works with any .stl I have imported.

See screenshot of G-code line 5816 where X and Y positions move into a previously printed object:

[image]

I do not see any place in this dialog box to insert or upload .stl and config files, but it appears that any .stl will do it (not sure about slic3r config, though.)

lordofhyphens, 2015-01-13:
There isn't a box for STL, you'll need to use some random file hosting site (I use 2shared myself for this sort of thing). The config is plain-text, so that can either go up zipped with the STL or pastebin.

alexrj, 2015-01-18:
@jeremyferris, I tried with two copies of a cube, using all default settings and just enabling Sequential Printing. As you can see from this rendering made with Repetier-Host, there's no offending move after the first object is completed:

[image]

You need to provide steps to reproduce the issue. You can copy and paste the contents of the configuration (File -> Export config) in a comment. If your tests show that the issue only exists with a specific STL file, you need to host the file somewhere and put a link here. Otherwise I'll test with cubes.

alexrj, 2015-01-18:
Uhm, by looking at my screenshot I see that there's something wrong. The downwards move shouldn't be in the center of the object but in the front-left corner, which is the origin (0,0) for that object. This STL cube is centered around the origin, so this means that with a model having very distant origin, the move would be positioned badly.
I fixed the issue. This is how it looks now:

[image]

No need to post any config or STL file. Thank you!

closed 2015-01-18 by alexrj