slic3r/Slic3r #2525: segfault when saving gcode from large stl files.

david-griffith, 2015-01-11:
Keep getting segfaults with large files when saving to gcode - built the current version in github, still the same. Small stl files don't seem to provoke the issue..... 

Running gdb on the process points to: 

    Program received signal SIGSEGV, Segmentation fault.
    [Switching to Thread 0x7fffc8d59700 (LWP 7144)]
    0x00007ffff51503f8 in Slic3r::ExtrusionEntityCollection::first_point (this=0x7fffb81218c0) at src/libslic3r/ExtrusionEntityCollection.cpp:50
    50	    return this->entities.front()->first_point();
    (gdb) bt
    #0  0x00007ffff51503f8 in Slic3r::ExtrusionEntityCollection::first_point (this=0x7fffb81218c0) at src/libslic3r/ExtrusionEntityCollection.cpp:50
    #1  0x00007ffff4fa56dc in XS_Slic3r__ExtrusionPath__Collection_first_point (my_perl=<optimised out>, cv=<optimised out>)
    at /usr/bin/perl -MExtUtils::XSpp::Cmd -e xspp -- -t "/home/david/Slic3r/xs/xsp/typemap.xspt"  "/home/david/Slic3r/xs/xsp/ExtrusionEntityCollection.xsp":71
    #2  0x00007ffff7b0a866 in Perl_pp_entersub () from /usr/lib/libperl.so.5.18
    #3  0x00007ffff7b02e86 in Perl_runops_standard () from /usr/lib/libperl.so.5.18
    #4  0x00007ffff7a94490 in Perl_call_sv () from /usr/lib/libperl.so.5.18
    #5  0x00007ffff5c2b4b9 in S_ithread_run (arg=0x4982020) at threads.xs:520
    #6  0x00007ffff6f6b182 in start_thread () from /lib/x86_64-linux-gnu/libpthread.so.0
    #7  0x00007ffff7785efd in clone () from /lib/x86_64-linux-gnu/libc.so.6

The function it's attempting to return appears to be working with an empty structure : 

    (gdb) p this->entities 
    $3 = {<std::_Vector_base<Slic3r::ExtrusionEntity*, std::allocator<Slic3r::ExtrusionEntity*> >> = {
    _M_impl = {<std::allocator<Slic3r::ExtrusionEntity*>> = {<__gnu_cxx::new_allocator<Slic3r::ExtrusionEntity*>> = {<No data fields>}, <No data fields>}, _M_start = 0x0, 
      _M_finish = 0x0, _M_end_of_storage = 0x0}}, <No data fields>}

And that's the extent of my debugging abilities :-)
I'll try and attach a file that provokes it.

alexrj, 2015-01-13:
Can you reproduce that with default settings?

kb-, 2015-01-13:
I could reproduce it with default settings on Win 8 x64.

david-griffith, 2015-01-13:
Tried it with the defaults after deleting my .slic3r directory, still the same.

alexrj, 2015-01-14:
Thank you, I fixed this.

closed 2015-01-14 by alexrj

curiouspl2, 2015-01-16:
i think i am getting it with 1.2.5 , waiting impatiently for 1.2.6 release to test it :)