slic3r/Slic3r #2526: 1.2.5-dev: OctoPrint send "Test" button returns HTTP 405 error, "send to printer" button in plater UI works

lordofhyphens, 2015-01-11:
Using Octoprint 1.1.1-14-g87ddcac and 1.2.5-dev, found that I was getting an HTTP 405 error (Method not allowed) when trying to "TEST". However, actually sending gcode to octoprint server works fine. 

The Bonjour discovery isn't working either, but I don't know if I can pin that one on slic3r. Well, it's detecting my Lexmark E260dn, but not my octoprint server.

foosel, 2015-01-12:
The "Test" button is `POST`ing to the `/api/version` resource instead of `GET`ing, which is the reason for this problem. See [here](https://github.com/alexrj/Slic3r/commit/24daa50bfd7e8402524900b8885e41697e670941#diff-7ff97188144bdd52c4dea49c7c907c52R1045)

alexrj, 2015-01-13:
Thank you, fixed by @lordofhyphens

closed 2015-01-13 by alexrj