slic3r/Slic3r #2527: --data-dir crashes slicer

repetier, 2015-01-11:
Repetier-Host uses --data-dir to allow different repository directories. This worked great so far, but testing 1.2.5, Windows, 64 bit it now crashes Slic3r with the error message:

09:22:26.289 : <Slic3r> variable is not of type Wx::Window at Z:/Users/littwin/Documents/Projekte/repetier/Repetier-Host/src/RepetierHost/bin/Release/Slic3r/lib/std/Slic3r/GUI.pm line 166.

Starting it without that parameter works fine.

Off topic wish:

with 1.2.2 you changed --solid-fill-pattern into  --external-fill-pattern. That makes it impossible to use newer versions with old Host versions since it now crashes. For the next release I have a switch to adjust changes on host side, but it is awful for users that do not know about these introduced incompatibilities and then get frustrated. Could you add --solid-fill-pattern as an alias, so older software can also use your slicer?

repetier, 2015-01-11:
A little update. It is not the --data-dir causing the problem. After copying the directory to original position I still had the problem. Deleting slic3r.ini helped. So it turned out it was this old version of slic3r.ini that caused the problems. All files are still there, so no idea why slic3r can not handle it's own config any more. Here the content if you want to check, if not just close the issue:

# generated by Slic3r 1.2.5 on Sun Jan 11 11:08:21 2015
autocenter = 1
background_processing = 1
last_output_path = C:\Users\littwin\AppData\Local\RepetierHost
last_version_check = 1407779027
mode = expert
version = 1.2.5

[presets]
filament = PLA ColorFabb 2.85 cool.ini
filament_1 = PLA 3mm - DD Extruder 0.5
filament_2 = - default -
print = 0.2 normal neu.ini
printer = DeltaTower Dual 0.4mm Düse xoff 20.ini

[recent]
config_directory = C:\Users\littwin
skein_directory = C:\Users\littwin\AppData\Local\RepetierHost

repetier, 2015-01-11:
Ok, now I know what the exact problem is. It is this line

printer = DeltaTower Dual 0.4mm Düse xoff 20.ini

You can not handle files with äöüÄÖÜ any more. As soon as I select it in printer Slic3r crashes. If I rename the file without spacial chars making it ASCII only, I can load the file. So I guess you have a problem with unicode handling in filenames, at least for windows where filenames are stored as UTF-16 and not UTF-8 like on mac and linux.

alexrj, 2015-01-13:
Thank you, I fixed handing of presets having Unicode characters and added `--solid-fill-pattern` as alias of `--external-fill-pattern`.

closed 2015-01-13 by alexrj