slic3r/Slic3r #2528: Crash in 1.2.5 on OSX slicing multiple objects

cdheiser, 2015-01-11:
Trying to slice all the objects that make up this part: http://www.thingiverse.com/thing:442063 at the same time results in a crash on OSX using 1.2.5

Crash details:

Process:         slic3r [12521]
Path:            /Applications/Slic3r.app/Contents/MacOS/slic3r
Identifier:      slic3r
Version:         Slic3r 1.2.5.92 (1.2.5.92)
Code Type:       X86 (Native)
Parent Process:  launchd [1857]
Responsible:     slic3r [12521]
User ID:         14378

Date/Time:       2015-01-11 09:12:16.889 -0800
OS Version:      Mac OS X 10.9.5 (13F34)
Report Version:  11
Anonymous UUID:  350BA24D-6435-C2B3-0241-61A6DAD8D2FD

Sleep/Wake UUID: A41CBF1A-5CFC-4ACA-8D8E-5B0EF596540D

Crashed Thread:  11

Exception Type:  EXC_BAD_ACCESS (SIGBUS)
Exception Codes: KERN_PROTECTION_FAILURE at 0x0000000000000000

VM Regions Near 0:
--> __PAGEZERO             0000000000000000-0000000000001000 [    4K] ---/--- SM=NUL  /Applications/Slic3r.app/Contents/MacOS/slic3r
    __TEXT                 0000000000001000-0000000000038000 [  220K] r-x/rwx SM=COW  /Applications/Slic3r.app/Contents/MacOS/slic3r

Thread 11 Crashed:
0   XS.bundle                     	0x05a128ec Slic3r::ExtrusionEntityCollection::first_point() const + 12
1   XS.bundle                     	0x0592a800 XS_Slic3r__ExtrusionPath__Collection_first_point(interpreter*, cv*) + 656
2   libperl.dylib                 	0x000e74a6 Perl_pp_entersub + 2294
3   libperl.dylib                 	0x000e5af6 Perl_runops_standard + 22
4   libperl.dylib                 	0x00074ca5 Perl_call_sv + 725
5   threads.bundle                	0x049dbfdd S_ithread_run + 1773
6   libsystem_pthread.dylib       	0x97a4a5fb _pthread_body + 144
7   libsystem_pthread.dylib       	0x97a4a485 _pthread_start + 130
8   libsystem_pthread.dylib       	0x97a4fcf2 thread_start + 34

Zuborg, 2015-01-12:
1.2.5 also crashes on more simple objects during generation of a G-code. We can see G-code unexpectedly terminated here. This is simple object tilted by 45 degrees before slicing.
[image]

alexrj, 2015-01-13:
Can you reproduce it also with default settings and without Repetier-Host?

cdheiser, 2015-01-14:
I just tried slicing the camera bed mount objects on 1.2.5 with all default settings and it crashed.  I don't use repetier-host.

dexy31337, 2015-01-14:
I've reproduced this problem on OS X 10.9, Slic3r 1.2.5, -default- settings. And got extra info.

The issue seems to be not with mutliple STLs, but only with arm.stl (and arm_flipped.stl, since it's generally same model). Slicing goes well (as a background process) and layer preview is ok, but Slic3r crashes on Export to G-code. On arm.stl crash is at layer Z37.55

Also I've checked versions 1.2.4 - 1.2.1, and found that they are unaffected by this behavior, so the problem is only with 1.2.5.

[Trace](http://pastebin.com/Gj4RTmnL)
[Incomplete g-code](https://dl.dropboxusercontent.com/u/3296729/arm.gcode)
This is layer, where it fails to export
[image],
And this is where it fails
[image]

alexrj, 2015-01-14:
Thank you, I fixed this.

closed 2015-01-14 by alexrj