slic3r/Slic3r #2530: OctoPrint integration: Port number from discovery not taken into account

foosel, 2015-01-12:
Slic3r version: 1.2.5
OS: Windows 7 64bit

When browsing for OctoPrint instances on the current network and selecting one of the found instances, the port number that is also part of the dns-sd record is not taken into account for prefilling the connection dialog and has to be added manually by the user (otherwise not connection will be possible). 

Example:

[image]

The instance in question is listening on port 5000, which can be verified using `dns-sd`:

    $ dns-sd -B _http._tcp
    Browsing for _http._tcp
    Timestamp     A/R Flags if Domain                    Service Type              Instance Name
    15:34:17.604  Add     3 11 local.                    _http._tcp.               OctoPrint instance "Development@Ripley"
    15:34:17.604  Add     3 28 local.                    _http._tcp.               OctoPrint instance "Development@Ripley"
    [...]
    
    $ dns-sd -L 'OctoPrint instance "Development@Ripley"' _http._tcp
    Lookup OctoPrint instance "Development@Ripley"._http._tcp.local
    15:35:56.426  OctoPrint\032instance\032"Development@Ripley"._http._tcp.local. can be reached at Ripley.local.:5000 (interface 11) Flags: 1
     path=/
    15:35:56.428  OctoPrint\032instance\032"Development@Ripley"._http._tcp.local. can be reached at Ripley.local.:5000 (interface 28)
     path=/

So the host field in slic3r's UI should actually be prefilled with "$host:$port", not just "$host", or alternatively an additional port field should be added. Manually adding the correct port fixes connectivity (although the Test button doesn't work due to #2526)

Have you thought about requiring a full base url here instead? That would also allow the usage of https and prefix URLs.

lordofhyphens, 2015-01-12:
I don't know how magical the underlying system is, so I can't say for sure if https would work right off the bat.

alexrj, 2015-01-13:
Thank you @foosel and @lordofhyphens. Parsing an optional protocol prefix in the configuration value is trivial, but supporting the protocol itself when it's not `http` isn't - I would need to package the SSL libraries along with Slic3r in order to support https, which is something that looks unneeded for now.

closed 2015-01-13 by alexrj