slic3r/Slic3r #2533: Unexpected boundary detection/bridging

sirkha, 2015-01-12:
I am experiencing some issues where on a couple of layers, Slic3r doesn't properly detect the edge of a part, and generates solid infill where there should be empty space.

Currently using Slic3r 1.2.5
Slic3r settings: https://github.com/sirkha/MotoX1ChargeCase/blob/master/Slic3rPrintSettings.ini
Part (Its an AMF with a modifier and some specific part settings): https://github.com/sirkha/MotoX1ChargeCase/blob/master/X1ChargeCase-inverted.amf

[image]

[image]
On this layer (at 10.50 mm), it fills the hole in the center of the image, when its not supposed to.
[image]
On the next layer, its fine.
[image]
The layer after that has the issue again, but all other layers are fine.

sirkha, 2015-01-12:
A couple of notes on the model and desired results:
Case_Body_inverted.stl is the main part that I am trying to get. I am using Partial Support to support a specific part of the end result. I have a dual-extruder setup, and will be printing Case_Body_inverted.stl using Ninjaflex, and everything else with HIPS, which I am going to dissolve away later. The modifier, Hex_Window.stl is setup so that its enclosed volume will not have a top or bottom infill, giving a see-thru hexagonal grid in that area.

I think the issue is how the modifier changes some of the perimeters. With just Case_Body_inverted.stl, i have no issues. As soon as I add the modifier to that (even without the support) i begin to see the issues I am having.

lordofhyphens, 2017-03-19:
Looks fixed to me by loading your amf in 1.3.0-dev 90395f93940410093244e9cfcaf3dce407ddc811
[image]

closed 2017-03-19 by lordofhyphens