slic3r/Slic3r #2534: Crash while exporting Gcode

kb-, 2015-01-12:
Slic3r is systematically crashing while exporting the Gcode generated from this [file](https://drive.google.com/file/d/0B3U2BO-9tuUCek1PdjFjVVgwRzQ/view?usp=sharing) (rotated 180 deg along X), with these [settings](https://drive.google.com/file/d/0B3U2BO-9tuUCM1FGRC1LZVN6bUk/view?usp=sharing).
The output is a partial [Gcode file](https://drive.google.com/file/d/0B3U2BO-9tuUCTkRWTEZWZW5Fams/view?usp=sharing).

Visual Studio gives the following error :
Unhandled exception at 0x00000000345FF7E8 (XS.dll) in slic3r.exe: 0xC0000005: Access violation reading location 0x0000000000000000.

I'm using version 1.2.5.35 on windows 8 (64). I have enough free memory.

david-griffith, 2015-01-12:
Hmm, sounds like the same thing as my issue under linux ( #2525 ). Can you see if it does the same thing with the file I linked to in that issue?

kb-, 2015-01-12:
You're right, it's crashing the same way

lordofhyphens, 2015-01-13:
I can reproduce your crash with Linux x64, newest dev release.

Here's the backtrace (ran gdb against perl with the coredump):
#0  0x00007f2a83ec0af8 in Slic3r::ExtrusionEntityCollection::first_point (this=0x7f2a400f3510) at src/libslic3r/ExtrusionEntityCollection.cpp:50
#1  0x00007f2a83d676a9 in XS_Slic3r__ExtrusionPath__Collection_first_point (my_perl=<optimized out>, cv=<optimized out>)
    at /usr/bin/perl -MExtUtils::XSpp::Cmd -e xspp -- -t "/home/lenox/slic3r-upstream/xs/xsp/typemap.xspt"  "/home/lenox/slic3r-upstream/xs/xsp/ExtrusionEntityCollection.xsp":71
#2  0x00007f2a8661a94b in Perl_pp_entersub () from /usr/lib/x86_64-linux-gnu/libperl.so.5.20
#3  0x00007f2a866131b6 in Perl_runops_standard () from /usr/lib/x86_64-linux-gnu/libperl.so.5.20
#4  0x00007f2a8659c4a5 in Perl_call_sv () from /usr/lib/x86_64-linux-gnu/libperl.so.5.20
#5  0x00007f2a84a14ae2 in S_ithread_run (arg=<optimized out>) at threads.xs:520
#6  0x00007f2a85e340a4 in start_thread (arg=0x7f2a718a3700) at pthread_create.c:309
#7  0x00007f2a85b68ccd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111

thiagopeixoto16, 2015-01-13:
Same here on Windows 7 x64 with 1.2.5 pre-compiled.

alexrj, 2015-01-14:
Thank you, I fixed this.

closed 2015-01-14 by alexrj

lordofhyphens, 2015-01-15:
Was the backtrace helpful at all?

alexrj, 2015-01-18:
Yes, thank you - it helped associating the error to the other (many) reports related to this stupid bug...