slic3r/Slic3r #2535: Segmentation Falt

fotis400, 2015-01-12:
I just downloaded the pre-built package of v1.2.5
It gives me a segmentation fault right after starting (before gui comes-in).

I'm using Fedora 21 x86_64.
Tell me if there are any logs to provide.

lordofhyphens, 2015-01-13:
Maybe the core dump will be helpful. Which package? x86 or x64?

lordofhyphens, 2015-01-13:
Well, I can confirm this on the x86_64 build (at least for me). I can't see a segfault on the x86 build.
Strange, because my from-source builds are just fine. 

here's  the backtrace from the core dump:
#0  0x00007f25328192d7 in wxWindow::DoSetSize(int, int, int, int, int) () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#1  0x00007f25329ad0ad in wxBoxSizer::RecalcSizes() () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#2  0x00007f25329df4cb in wxWindowBase::Layout() () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#3  0x00007f25329df696 in wxWindowBase::InternalOnSize(wxSizeEvent&) () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#4  0x00007f25322fbdf0 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
   from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#5  0x00007f25322fbfaf in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#6  0x00007f25322fc02d in wxEvtHandler::TryHereOnly(wxEvent&) () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#7  0x00007f25322fc91d in wxEvtHandler::ProcessEventLocally(wxEvent&) () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#8  0x00007f25322fc974 in wxEvtHandler::ProcessEvent(wxEvent&) () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#9  0x00007f25322fc0c4 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#10 0x00007f25329dedf0 in wxWindowBase::HandleWindowEvent(wxEvent&) const () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#11 0x00007f2532819460 in wxWindow::DoSetSize(int, int, int, int, int) () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#12 0x00007f2532817c37 in wxWindow::DoSetClientSize(int, int) () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#13 0x00007f25329ad613 in wxSizer::SetSizeHints(wxWindow*) () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#14 0x00007f25334d5044 in XS_Wx__Sizer_SetSizeHints(interpreter*, cv*) ()
   from /home/lenox/slicer-1.2.5/Slic3r/lib/vrt/8b57d08e0d55cfe6498642b59ecf5ad1/Wx.so
#15 0x00007f2537606db3 in Perl_pp_entersub () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libperl.so
#16 0x00007f25376053fe in Perl_runops_standard () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libperl.so
#17 0x00007f253759f28b in Perl_call_sv () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libperl.so
#18 0x00007f25332da889 in XS_Wx___App_Start(interpreter*, cv*) () from /home/lenox/slicer-1.2.5/Slic3r/lib/vrt/8b57d08e0d55cfe6498642b59ecf5ad1/Wx.so
#19 0x00007f2537606db3 in Perl_pp_entersub () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libperl.so
#20 0x00007f25376053fe in Perl_runops_standard () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libperl.so
#21 0x00007f253759fba3 in perl_run () from /home/lenox/slicer-1.2.5/Slic3r/bin/../dll/libperl.so
#22 0x0000000000403064 in ?? ()
#23 0x00007f253645db45 in __libc_start_main (main=0x402df0, argc=1, argv=0x7fff0034dc38, init=<optimized out>, fini=<optimized out>, 
    rtld_fini=<optimized out>, stack_end=0x7fff0034dc28) at libc-start.c:287
#24 0x0000000000401b69 in ?? ()
#25 0x00007fff0034dc28 in ?? ()
#26 0x000000000000001c in ?? ()
#27 0x0000000000000001 in ?? ()
#28 0x00007fff0034eb1d in ?? ()
#29 0x0000000000000000 in ?? ()

Patola, 2015-01-13:
It segfaults at my face too, but I just git pulled and compiled on my laptop (Ubuntu 14.10 64 bits, 16 gig memory) and it starts the graphical interface. *But* it stops slicing at even the most trivial stl, with error "Caugh C++ exception of type or derived from 'std::exception': vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 25) at /dados"

lordofhyphens, 2015-01-13:
@Patola I'm not seeing this behavior with my own copy of recent pull with my settings. Could you post another issue with an STL and a copy of the settings you are using?

Patola, 2015-01-13:
@lordofhyphens, sure, but this bug report pertains to a coredump. Should I open a new bug for the slicing problem? What is the debug information I need to include, besides the slicer settings and the stl file?

Patola, 2015-01-13:
I opened a new report (2537) for my issue, let's not confuse stuff. Sorry for polluting this one.

lordofhyphens, 2015-01-13:
Well, I rotated the provided STL 90deg Y, and then cut it. The top section ended up in the air and the lower section ended up below the bed.

I exported the STL, added it again to the plater and split, which put it back correctly on the plater. And it sliced and exported gcode without crashing when I did that.

lordofhyphens, 2015-01-13:
well, nevermind then--just realized I was running from my source build. Time to figure out which commit broke things :D

lordofhyphens, 2015-01-13:
Stepping around through the source with git-bisect, looks like the problem is possibly
8f4cbefd0dd71cd6ae8dc149bd2e349905cd1ee8 or 
39172d5a08f22aaa660ec2d836c2857ef6383777. I'll take another pass at bisect later to make sure that I'm feeding it the right info. Is there a clean switch for the build so I can delete previously installed XS modules to ensure I don't have any pollution?

Steps were to build slic3r, and then load the STL, rotate it and export gcode. During the bisect I am ignoring failed tests when selecting good/bad if the XS finished compilation and installed.

lordofhyphens, 2015-01-13:
Unfortunately, I can't test 8f4cbefd0dd71cd6ae8dc149bd2e349905cd1ee8 in isolation, but 39172d5a08f22aaa660ec2d836c2857ef6383777 is a trivial commit.

Hope this helps.

alexrj, 2015-02-01:
I was unable to reproduce this, actually. But I haven't put much effort (just lack of time!). In the mean time I released 1.2.6. @fotis400, I'd appreciate if you could confirm whether this is fixed as of 1.2.6 or still exists. Thank you!

oschonrock, 2015-02-01:
sorry, still here for me:

Slic3r/bin$ ./slic3r 
Segmentation fault (core dumped)

uname -a
Linux XPS-8300 3.13.0-24-generic #47-Ubuntu SMP Fri May 2 23:30:00 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux

fotis400, 2015-02-02:
I still get the SEGFAULT too. 

It must be stated that only the build provided causes this. Building from sources works excellent for me.
Where can I find the core dump?

lordofhyphens, 2015-02-02:
Usually a core dump for Linux gets put in the direction you ran it from.

For Windows i think that there is useful info in the crash log in the event
log.

fotis400, 2015-02-02:
Of course I checked already on the same directory, and there isn't any file there. I also checked in a few common places, but no luck. So I am asking whether the programmer intentionally has changed the path of the dump, (if it is possible...) or any other clue...

jamesgao, 2015-02-03:
If this is the same error as I'm seeing, here's a gdb trace from the segfault:

```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff29f32d7 in wxWindow::DoSetSize(int, int, int, int, int) () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
(gdb) bt
#0  0x00007ffff29f32d7 in wxWindow::DoSetSize(int, int, int, int, int) () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#1  0x00007ffff2b870ad in wxBoxSizer::RecalcSizes() () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#2  0x00007ffff2bb94cb in wxWindowBase::Layout() () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#3  0x00007ffff2bb9696 in wxWindowBase::InternalOnSize(wxSizeEvent&) () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#4  0x00007ffff24d5df0 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
   from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#5  0x00007ffff24d5faf in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#6  0x00007ffff24d602d in wxEvtHandler::TryHereOnly(wxEvent&) () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#7  0x00007ffff24d691d in wxEvtHandler::ProcessEventLocally(wxEvent&) () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#8  0x00007ffff24d6974 in wxEvtHandler::ProcessEvent(wxEvent&) () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#9  0x00007ffff24d60c4 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#10 0x00007ffff2bb8df0 in wxWindowBase::HandleWindowEvent(wxEvent&) const () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#11 0x00007ffff29f3460 in wxWindow::DoSetSize(int, int, int, int, int) () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#12 0x00007ffff29f1c37 in wxWindow::DoSetClientSize(int, int) () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#13 0x00007ffff2b87613 in wxSizer::SetSizeHints(wxWindow*) () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#14 0x00007ffff36af044 in XS_Wx__Sizer_SetSizeHints(interpreter*, cv*) () from /auto/k1/james/Dropbox/prusa/Slic3r/lib/vrt/8b57d08e0d55cfe6498642b59ecf5ad1/Wx.so
#15 0x00007ffff7b15db3 in Perl_pp_entersub () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libperl.so
#16 0x00007ffff7b143fe in Perl_runops_standard () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libperl.so
#17 0x00007ffff7aae28b in Perl_call_sv () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libperl.so
#18 0x00007ffff34b4889 in XS_Wx___App_Start(interpreter*, cv*) () from /auto/k1/james/Dropbox/prusa/Slic3r/lib/vrt/8b57d08e0d55cfe6498642b59ecf5ad1/Wx.so
#19 0x00007ffff7b15db3 in Perl_pp_entersub () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libperl.so
#20 0x00007ffff7b143fe in Perl_runops_standard () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libperl.so
#21 0x00007ffff7aaeba3 in perl_run () from /auto/k1/james/Dropbox/prusa/Slic3r/bin/../dll/libperl.so
#22 0x0000000000403064 in ?? ()
#23 0x00007ffff6945ec5 in __libc_start_main (main=0x402df0, argc=1, argv=0x7fffffffde28, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, 
    stack_end=0x7fffffffde18) at libc-start.c:287
#24 0x0000000000401b69 in ?? ()
#25 0x00007fffffffde18 in ?? ()
#26 0x000000000000001c in ?? ()
#27 0x0000000000000001 in ?? ()
#28 0x00007fffffffe1bb in ?? ()
#29 0x0000000000000000 in ?? ()
```

ccavanaugh, 2015-02-05:
More or less the same issue on 64 bit Arch Linux install

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff2d172d7 in wxWindow::DoSetSize(int, int, int, int, int) () from /home/craig/bin/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
(gdb) bt
#0  0x00007ffff2d172d7 in wxWindow::DoSetSize(int, int, int, int, int) () from /home/craig/bin/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#1  0x00007ffff2eab0ad in wxBoxSizer::RecalcSizes() () from /home/craig/bin/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#2  0x00007ffff2edd4cb in wxWindowBase::Layout() () from /home/craig/bin/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#3  0x00007ffff2edd696 in wxWindowBase::InternalOnSize(wxSizeEvent&) () from /home/craig/bin/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#4  0x00007ffff27f9df0 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
   from /home/craig/bin/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#5  0x00007ffff27f9faf in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) () from /home/craig/bin/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#6  0x00007ffff27fa02d in wxEvtHandler::TryHereOnly(wxEvent&) () from /home/craig/bin/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#7  0x00007ffff27fa91d in wxEvtHandler::ProcessEventLocally(wxEvent&) () from /home/craig/bin/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#8  0x00007ffff27fa974 in wxEvtHandler::ProcessEvent(wxEvent&) () from /home/craig/bin/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#9  0x00007ffff27fa0c4 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () from /home/craig/bin/Slic3r/bin/../dll/libwx_baseu-2.9.so.3
#10 0x00007ffff2edcdf0 in wxWindowBase::HandleWindowEvent(wxEvent&) const () from /home/craig/bin/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#11 0x00007ffff2d17460 in wxWindow::DoSetSize(int, int, int, int, int) () from /home/craig/bin/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#12 0x00007ffff2d15c37 in wxWindow::DoSetClientSize(int, int) () from /home/craig/bin/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#13 0x00007ffff2eab613 in wxSizer::SetSizeHints(wxWindow*) () from /home/craig/bin/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#14 0x00007ffff39d3044 in XS_Wx__Sizer_SetSizeHints(interpreter*, cv*) () from /home/craig/bin/Slic3r/lib/vrt/8b57d08e0d55cfe6498642b59ecf5ad1/Wx.so
#15 0x00007ffff7b16db3 in Perl_pp_entersub () from /home/craig/bin/Slic3r/bin/../dll/libperl.so
#16 0x00007ffff7b153fe in Perl_runops_standard () from /home/craig/bin/Slic3r/bin/../dll/libperl.so
#17 0x00007ffff7aaf28b in Perl_call_sv () from /home/craig/bin/Slic3r/bin/../dll/libperl.so
#18 0x00007ffff37d8889 in XS_Wx___App_Start(interpreter*, cv*) () from /home/craig/bin/Slic3r/lib/vrt/8b57d08e0d55cfe6498642b59ecf5ad1/Wx.so
#19 0x00007ffff7b16db3 in Perl_pp_entersub () from /home/craig/bin/Slic3r/bin/../dll/libperl.so
#20 0x00007ffff7b153fe in Perl_runops_standard () from /home/craig/bin/Slic3r/bin/../dll/libperl.so
#21 0x00007ffff7aafba3 in perl_run () from /home/craig/bin/Slic3r/bin/../dll/libperl.so
#22 0x0000000000403064 in ?? ()
#23 0x00007ffff696e040 in __libc_start_main () from /usr/lib/libc.so.6
#24 0x0000000000401b69 in ?? ()
#25 0x00007fffffffe398 in ?? ()
#26 0x000000000000001c in ?? ()
#27 0x0000000000000001 in ?? ()
#28 0x00007fffffffe70c in ?? ()
#29 0x0000000000000000 in ?? ()
(gdb)

alexrj, 2015-02-09:
I really can't reproduce this one, but all of your stacktraces look very similar so a problem exists indeed.
I wonder whether this might be caused by the version check bug that I fixed in 4083b3380712b1a3ab6f799c5652c22669667653 today.  Infact, the packaged binary performs automatic version check while the source builds don't.

alexrj, 2015-02-09:
Could you try to disable automatic version check in `~/.Slic3r/slic3r.ini` by setting `version_check = 0`?

oschonrock, 2015-02-09:
tried that just now, sorry no joy.

oliver@oliver-XPS-8300:~/Downloads/Slic3r-1-2-6-x64$ cat ~/.Slic3r/slic3r.ini 
# generated by Slic3r 1.2.6 on Mon Feb  9 14:20:24 2015
autocenter = 1
background_processing = 1
mode = expert
version = 1.2.6
version_check = 0

[presets]
filament = 
print = 
printer = 

oliver@oliver-XPS-8300:~/Downloads/Slic3r-1-2-6-x64$ bin/slic3r 
Segmentation fault (core dumped)

alexrj, 2015-02-09:
@oschonrock, thank you for trying that.

SkyRzn, 2015-02-10:
I got segfault with same backtrace. Gentoo x86_64.
Slic3r successfully launched and works with --no-plater option.

oschonrock, 2015-02-10:
confirmed...no segfault with --no-plater. i guess that narrows it down "somewhat"

alexrj, 2015-02-10:
Yeah, thank you very much for the --no-plater hint!  Narrows it down to some GUI element in the plater tab. Maybe the sliders in the 2D/3D previews.  But I need to reproduce this in order to fix it. Damn!

alexrj, 2015-02-10:
Oh, could you run it with --debug please and let me know the wxWidgets version?

SkyRzn, 2015-02-10:
wxWidgets version wxWidgets 2.9.3, Wx version 0.9906

oschonrock, 2015-02-10:
identical here.
wxWidgets version wxWidgets 2.9.3, Wx version 0.9906

ccavanaugh, 2015-02-10:
wxWidgets version wxWidgets 2.8.12, Wx version 0.9923 on Arch 64

insomniacUNDERSCORElemon, 2015-02-11:
Same issue here on Arch, oddly though Repetier has no issue using it (I first noticed when I saw the overlap feature) so it is very likely a problem with the GUI like others are saying.

SkyRzn, 2015-02-11:
It seems that the issue does not depend on the wxWidgets version. I tried Slic3r with the plater on another system and it works without problems.
Ubuntu 14.04.1 LTS, x86_64.
wxWidgets version wxWidgets 2.9.3, Wx version 0.9906

oschonrock, 2015-02-11:
Interesting: I have 
Ubuntu 14.04 LTS _Trusty Tahr_ - Release amd64 (20140417)]

bin/slic3r  --debug
wxWidgets version wxWidgets 2.9.3, Wx version 0.9906
Data directory: /home/oliver/.Slic3r
Segmentation fault (core dumped)

subtle package version differences?
I have a couple of ppas installed on my system freecad-daily openscad/releases etc, to get more up to date versions of those. Those are installed on most of the systems which I have seen this Segfault on. Could the ppas be updating some lib that causes the segfault? I will see if I can find a clean Ubuntu x64 machine to try on.

oschonrock, 2015-02-11:
nope..that's a dead end. tried on 14.04 (not 14.04.1) machine without those ppas. still segfaults.

Could this be video driver related? I am using nvidia throughout usually this:
NVIDIA binary driver - version 331.113

SkyRzn, 2015-02-11:
Hmmm... May be you are right.
I have a nvidia card on my home system, where a got the segfault.
On my office computer I have an AMD graphics card and there is all ok.

SkyRzn, 2015-02-11:
There was a similar issue in Hugin: https://bugs.launchpad.net/hugin/+bug/679427

alexrj, 2015-02-11:
@SkyRzn, could be related indeed. Maybe http://wiki.panotools.org/Hugin_FAQ#Hugin_Quits_.28Seg_Faults.29_at_Launch_.28Linux.29 work here too?

I need to set up a non-Ubuntu box and try myself. If I can get it to reproduce, I can easily fix it.

@ccavanaugh, so I guess you're running from git and still getting this error?

SkyRzn, 2015-02-11:
@alexrj glxgears succefully runs on my system. Maybe disabling desktop effects will solve the problem. I will try it.

ccavanaugh, 2015-02-12:
Yes, running the git build that Arch has in the repository.

SkyRzn, 2015-02-12:
No, desktop effects disabling does not work. I still get segfaults.

zennmaster, 2015-02-25:
Any progress on this? Still having issues with segfaults at startup on Ubuntu 14.04.

AndKe, 2015-03-01:
same issue here with 1.2.6 on Ubuntu 14.10 + Nvidia.
release 1.2.0 works fine on this system.

brackendawson, 2015-03-10:
My work laptop sees this on opening Slic3r GUI, my home laptop is fine:

Work laptop: Lenovo W510 w/ nVidia Quadro FX 880M running Ubuntu 14.04 64bit.
Home laptop: Lenovo X121e w/ Intel graphics running Ubuntu 14.10 64-bit.

thehans, 2015-03-13:
Also seeing segfault here with slic3r 1.2.6, same Lenovo W510 laptop as brackendawson, on Ubuntu 14.10 64bit, cinnamon desktop.

thehans, 2015-03-13:
Also tested same config as above with Nvidia driver 304.125
still segfaults

thehans, 2015-03-13:
tested all the 1.2.x 64bit releases to confirm that only 1.2.5 and later crashes

1.2.4 and earlier do not segfault

bcolsen, 2015-03-16:
I have the same problem on Ubuntu 14.04 running Nvidia 340.65 drivers

vetterfl, 2015-03-22:
I get the same error after changeing to a Nvidia card and using propritairy drivers (340.76). Maybe that helps.

dumle29, 2015-04-02:
Seems it's a problem with NVidia drivers:
Same issue. Ubuntu 14.10 64bit, GTX660 with Nvidia 331.113 propritairy driver.

lordofhyphens, 2015-04-02:
That driver is pretty old now, have you tried the newer versions of the
proprietary drivers?

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

zennmaster, 2015-04-02:
I'm getting the same behavior with 346.47.

lordofhyphens, 2015-04-02:
I'm not  getting the segfault on launch with 346.47 on a GT740M.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

zennmaster, 2015-04-02:
Lordofhyphens, Are you running multiple monitors?

AndKe, 2015-04-03:
I think we are onto something here. Both at home, and work, it's my multi-monitor workstations that fail.
Another important clue; no other application fails in a similar way, and I do use many applications with opengl or otherwise comparable.

SkyRzn, 2015-04-03:
I got same problems with one monitor. I will try different drivers.

orestesmas, 2015-04-03:
Got same problems, also with single monitor.
    slic3r 1.2.6
    nvidia driver 349.12 (from ubuntu packages)
    ubuntu 14.10 with linux kernel 3.16.0-33-generic x86_64

Strangely, a similar configuration worked fine previously for me, but I changed the nvidia module (and later used unofficial xorg packages from ppa:xorg-edgers) and slic3r started to give segmentation faults.

The working combination was:
    * Standard xorg packages
    * nvidia driver "NVIDIA-Linux-x86_64-346.47" manually installed from vendor bundle

After this post, I'll try to revert to my working combination, and post the results.

pcwalden, 2015-04-04:
Same problem with Fedora 21 and older NVIDIA NV05  graphics card using nouveau driver.

oxyum, 2015-04-11:
Guys,

I just removed libGL.so.\*/libGLU.so.\*/libglut.so.\* binaries from /dll/ catalog and Slic3r v1.2.6 not crashed.

PS: Ubuntu 14.04 x86_64, Nvidia GTX570, driver version 331.113.

AndKe, 2015-04-11:
@oxyum   - Thank you.
Confirmed, removing those pesky links/libs works great.
Also, my last working version, 1.2.0  - did not had these files.

- so... what is the point of having them there ?

lordofhyphens, 2015-04-11:
Odds are to avoid a package dependency on another platform.

ccavanaugh, 2015-04-26:
Confirmed on Arch that removal of the noted libs fixes the crash.

brackendawson, 2015-04-27:
Another confirmation here on Ubuntu 14.04 x86_64 on my nVidia machine.

[5 more replies]

closed 2015-06-09 by alexrj