slic3r/Slic3r #2537: Slicing stops at trivial STL, C++ exception of type

Patola, 2015-01-13:
When I try to slice even the simplest STLs, slic3r stops the slicing with this message:

Caugh C++ exception of type or derived from 'std::exception': vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 25) at /dados

Patola, 2015-01-13:
stl file I used: https://drive.google.com/file/d/0B8WZj6JJHZDwRnZQRkRpYVAwWlk/view?usp=sharing

slic3r settings, in zip format: https://drive.google.com/file/d/0B8WZj6JJHZDwbWdVdm1yTmw4Mlk/view?usp=sharing

Patola, 2015-01-13:
Please tell me how I can add any further debug information.

lordofhyphens, 2015-01-13:
Well, you could run "gdb perl /path/to/corefile" and then type "back" to get the backtrace. It might be useful if you compiled from source.

Patola, 2015-01-13:
It is not dumping core, it is showing the message on the bottom window border...

Patola, 2015-01-13:
[image]

Patola, 2015-01-13:
I even found out a way to avoid this problem. I don't know how, but using my settings:
- Select Graber i3 for print settings and Graber i3 for Printer.
- Let is slice and finish.
- Select Sethi3D_no_support for print settings and Sethi3D_Printer_Settings for printer.

Voilà. I don't know why but the slicing error stops. From now on, every stl is sliced correctly.

Using diff to compare the settings:
print settings -
most notable differences between sethi3d (1) and graber i3 (2):
- (1) no support, (2) with support
- (1) seam position random, (2) neareest

Others are numeric differences which I reckon are not likely to be the cause.

For printer settings:
Sethi3D has these codes, graber i3 does not:
octoprint_apikey = 
octoprint_host = 
pressure_advance = 0

alexrj, 2015-01-18:
Thank you, I was able to reproduce and fix this. It was caused by `skirts` being > 0 but `skirt_height` being 0. I also included a regression test.

closed 2015-01-18 by alexrj