slic3r/Slic3r #2538: 1.2.5 exp - exporting takes a lot of time when "avoid crossing perimeters" is checked

gege2b, 2015-01-13:
Hi there

May be it's not bug, may be it's only esthetics, I don't know. In this case sorry for this report.

It works with any STL, but here is the one I'm using : https://dl.dropboxusercontent.com/u/34934879/bastia%20logo%20creux%2010x10.STL

The config is not important, except for "avoid crossing perimeters". When checked, Slic3r takes a LOT of time on the "exporting gcode..." status.

I know this option is time consuming, but I trought this should occurs on the "generating perimeters" phase, not on the exporting one

cheers

dexy31337, 2015-01-14:
I've noticed, that "exporting gcode" phase is most time-consuming for difficult models, especially with small layers (0.1mm). While slicing is multithreaded, I believe that expoting g-code is sequential as file is written line-by-line. So, due to great amount of lines (hundreds of thousands), it is normal, that exporting takes some time.
Same with "avoid crossing perimeters", as it generates a lot of extra movement commands.

lordofhyphens, 2016-07-11:
Try current 1.3.0-dev (see link on main readme), reopen if there are still issues. Avoid crossing perimeters is a bottleneck in the code output.

closed 2016-07-11 by lordofhyphens