slic3r/Slic3r #2539: Slic3r randomly produces layers with no infill

triffid, 2015-01-14:
Using d1243397fa17e0c4ace3f62fc0d0937ce89ab153 (Wed Jan 14 00:14:56 2015 +0100)

STL: http://triffid-hunter.no-ip.info/sprockets.stl
config: http://triffid-hunter.no-ip.info/pla_95rect.ini
gcode: http://triffid-hunter.no-ip.info/sprockets_PLA_EW0.5_LH0.2_ND0.35_FD95%.gcode
image: http://triffid-hunter.no-ip.info/sprockets.png

also, what's going on with the travel moves?

triffid, 2015-01-14:
Just found out, if I change all the extrusion widths from 0.5 to 0.481, slic3r segfaults

triffid, 2015-01-14:
0.55 widths segfault, 0.51 widths slice but give a zero infill layer at 2.6mm (compared to 2.4mm and 8.8mm in first post)

alexrj, 2015-01-15:
The segfault should be fixed as of bf02062a679e3ef0cf16aa285817cd9022ee500c.

However I can't reproduce the missing infill layers with HEAD or the commit you specified. Also, I can't download the G-code file from your server, maybe broken link...

alexrj, 2015-01-15:
The correct link is:
http://triffid-hunter.no-ip.info/sprockets_PLA_EW0.5_LH0.2_ND0.35_FD95%25.gcode

alexrj, 2015-01-16:
I could reproduce the issue. It's a Clipper bug and I managed to isolate a failing test case.

triffid, 2015-01-17:
yaay! reproducibility is wonderful, lovely when it's not just me :)

alexrj, 2015-01-18:
This is the upstream report:
https://sourceforge.net/p/polyclipping/bugs/123/

alexrj, 2015-01-30:
Well, I pushed a workaround. It works now.

closed 2015-01-30 by alexrj