slic3r/Slic3r #2540: [feature request] number of copies

elkooo, 2015-01-14:
Hey there, 

I often have to print multiple copies of small, complex structures. It is very time consuming if I need to press the «+» button 30 times and have to wait in between the klicks until the plater is loaded. Instead it would be great if I could directly type the number of copies for example by right-clicking it. Instead of just increasing / decreasing one by one, a further option «set quantity...» or «number of copies…» to manually set the number by typing it in a small popup dialog or so would be amazing!
Don't know if this is possible though…

[image]

dexy31337, 2015-01-14:
Good idea. I'd allow to enter number of copies directly into right table. Here
[image]

alexrj, 2015-01-14:
That's a good idea @elkooo, thank you. I added a command for that.

closed 2015-01-14 by alexrj

brotherchris, 2017-11-01:
What would the command on command line be for doing something like this?

lordofhyphens, 2017-11-01:
@brotherchris `./slic3r.pl --help` output: 

````
    --duplicate         Number of items with auto-arrange (1+, default: 1)
    --duplicate-grid    Number of items with grid arrangement (default: 1,1)
    --duplicate-distance Distance in mm between copies (default: 6)
````

brotherchris, 2017-11-01:
Thanks, I have tried all these and can't seem to make them work. The other
commands work, just not duplicate. Can someone test and see if it works for
them?