slic3r/Slic3r #2542: Gaps near edges of rectilinear infill

Vicious-one, 2015-01-14:
https://github.com/alexrj/Slic3r/commit/04aa240265ea35d881dd5d395255bf9db23fcb52
This commit causes visible gaps when infill is parallel to perimeter and gap fill is not triggered either.

[image]
[image]

alexrj, 2015-01-15:
@Vicious-one, is the first screenshot showing a 100% density layer or an internal infill with high density (but not 100%)?

Vicious-one, 2015-01-15:
First layer is 100% solid (bottom screenshot, short gaps closer to the top), infill is at 90% (top screenshot, long gaps near the sides).

http://vserv.sinp.msu.ru/temp/gap-rect.tar.gz

alexrj, 2015-01-15:
Internal infill is not adapted to fit the entire space when density is not 100%, so it's normal that there can be gaps.  If you don't want gaps, set it to 100%.

Vicious-one, 2015-01-15:
What bothers me is that it happens even when infill is 100% solid. For example, this bottom layer:

Current version:
[image]

Earlier:
[image]

a4jp-com, 2015-01-16:
You could tune it using the advanced settings and change the size of the infill. For example, 0.????.

Vicious-one, 2015-01-16:
Changing path width to fix one layer will inevitably make a mess somewhere else. That is not a solution.
My point is that maybe infill overlap tolerance, infill spacing or gap fill triggering may need to be adjusted.

alexrj, 2015-02-01:
@Vicious-one, yeah, that's a tricky case. For solid layers, Slic3r *does* adapt toolpaths and their flow in order to fill the entire space without leaving side gaps. This happens also in your last screenshot (look at the extreme parts at the sides), but then there's a problem when the object contour has other "steps" in intermediate positions which is where you see the gaps… Not sure I was clear enough :)

alexrj, 2015-02-01:
Anyway, I partially reverted that commit and made a configurable option for perimeter/infill overlap, so I think this can be closed.

closed 2015-02-01 by alexrj