slic3r/Slic3r #2543: Wrong value for first_layer_extrusion_width in cli

adius, 2015-01-15:
When using slic3r like this:
```
$ ~/Applications/Slic3r.app/Contents/MacOS/slic3r --output out.gcode in.stl
```
I get a gcode file with the comment `; first_layer_extrusion_width = 200%`.

When I use it like this however:
```
$ ~/Applications/Slic3r.app/Contents/MacOS/slic3r --output out.gcode --first-layer-extrusion-width 0 out.stl
```
I get the comment `; first_layer_extrusion_width = 0`
I'm not quite sure where exactly the error comes from after glancing at the code.

alexrj, 2015-01-15:
You assume that's an "error" and it's "wrong" but actually you're wrong and I can't see what would the error be.

200% is the default value.

closed 2015-01-15 by alexrj

adius, 2015-01-15:
When i set `--first-layer-extrusion-width 0` its supposed to use the automatically set 200%.
In the gcode file however it writes that it used a first_layer_extrusion_width of 0 !?

alexrj, 2015-01-15:
"it's supposed to use the automatically set 200%"? Not really, where do you read that?

adius, 2015-01-15:
https://github.com/alexrj/Slic3r/blob/9a7e5327abbdbdfdaf8a0df6ecb84b8f9849bf16/xs/src/libslic3r/PrintConfig.cpp#L322 : "leave 0 for default"

So when i set '--first-layer-extrusion-width 0' it should use 200%

alexrj, 2015-01-15:
The default extrusion width is the one set using the "Default extrusion width" option:

https://github.com/alexrj/Slic3r/blob/9a7e5327abbdbdfdaf8a0df6ecb84b8f9849bf16/xs/src/libslic3r/PrintConfig.cpp#L206

adius, 2015-01-15:
Ah OK, got it. Very confusing. I guess we should clarify the command line documentation. Maybe something like:

```
--first-layer-extrusion-width
Set a different extrusion width for first layer (default: 200%).
Set to 0 to use the same extrusion width as specified at --extrusion-width.
```

alexrj, 2015-01-18:
I clarified the tooltip.