slic3r/Slic3r #2544: Support material Skirt issues

sirkha, 2015-01-15:
Problem 1: When generating automatic support material, some times a layer just for support material is generated in between the normal layers of the print. I am not sure if this is by design, but either way, the skirt is not on that layer in the tool path preview, but in the G-code it is. The problems with this are two-fold: the skirt basically tries to print twice in the same place which for me has resulted in skirts getting torn off the bed, and also has caused the extruder to jam, as too much back pressure is created and the hobbed bolt wears a hole in the fillament.

Problem 2: Tall skirts don't like to stay on the bed for the whole print. This is caused by two things, I think. First, the skirt is very thin on the bottom unless you a large number of loops, and even then, with ABS-like materials, this results in shrinkage pulling it off the bed. To solve this, I would implement a first-layer skirt loops option that effectively gives the skirt a brim. The second issue with skirt-bed adhesion I am experiencing is with the multi-material first layer skirt. I understand the want/need to build pressure by generating a skirt loop with the second extruder on the first layer, but when using a different material, like Ninjaflex, in my case, the second layer skirt doesn't want to stick to it very well. I would solve this by always putting the non-skirt-extruder material in an additional loop outside where the skirt will print on the next layer.

sirkha, 2015-01-15:
I am having both these problems with slic3r version 1.2.0 through 1.2.5:

alexrj, 2015-01-15:
In the guidelines it's clearly written that multiple distinct things should not be mixed in a single report.
It's also written that you need to provide some required things in order to reproduce the issues. Can you please read the guidelines and adapt this one? Otherwise it's very unlikely that your problems get fixed.

sirkha, 2015-01-16:
Config file used
=========
    # generated by Slic3r 1.2.5 on Thu Jan 15 18:42:20 2015
    avoid_crossing_perimeters = 0
    bed_shape = 0x0,300x0,300x300,0x300
    bed_temperature = 90
    bottom_solid_layers = 3
    bridge_acceleration = 0
    bridge_fan_speed = 0
    bridge_flow_ratio = 1.5
    bridge_speed = 18
    brim_width = 0
    complete_objects = 0
    cooling = 1
    default_acceleration = 0
    disable_fan_first_layers = 1
    dont_support_bridges = 1
    duplicate_distance = 6
    end_gcode = T0 ; Select 1st extruder\nM104 S0 ; turn off E0 nozzle temperature\nT1 ; Select 2nd extruder\nM104 S0 ; Turn off E1 nozzle temperature\nM140 S0 ; Turn off bed temp\nG91 ; Relative Move\nG1 Z5 F5000 ;\nG90 ; Absolute Move\nG28 X0  ; home X axis\nG1 Y305 F10800.000; move tray out\nM84     ; disable motors\nT0    ; return to default
    external_fill_pattern = rectilinear
    external_perimeter_extrusion_width = 160%
    external_perimeter_speed = 50%
    external_perimeters_first = 0
    extra_perimeters = 1
    extruder_clearance_height = 20
    extruder_clearance_radius = 50
    extruder_offset = 0x0,0x-53.6
    extrusion_axis = E
    extrusion_multiplier = 1,1
    extrusion_width = 0.4
    fan_always_on = 0
    fan_below_layer_time = 60
    filament_diameter = 2.9,2.95
    fill_angle = 30
    fill_density = 40%
    fill_pattern = honeycomb
    first_layer_acceleration = 0
    first_layer_bed_temperature = 84
    first_layer_extrusion_width = 0.5
    first_layer_height = 0.3
    first_layer_speed = 50%
    first_layer_temperature = 230,230
    gap_fill_speed = 40
    gcode_arcs = 0
    gcode_comments = 0
    gcode_flavor = reprap
    infill_acceleration = 0
    infill_every_layers = 1
    infill_extruder = 2
    infill_extrusion_width = 0.6
    infill_first = 0
    infill_only_where_needed = 0
    infill_speed = 90
    interface_shells = 0
    layer_gcode = 
    layer_height = 0.15
    max_fan_speed = 0
    min_fan_speed = 0
    min_print_speed = 5
    min_skirt_length = 0
    notes = 
    nozzle_diameter = 0.35,0.35
    octoprint_apikey = 
    octoprint_host = 
    only_retract_when_crossing_perimeters = 1
    ooze_prevention = 1
    output_filename_format = [input_filename_base].gcode
    overhangs = 1
    perimeter_acceleration = 0
    perimeter_extruder = 2
    perimeter_extrusion_width = 0.38
    perimeter_speed = 80
    perimeters = 3
    post_process = 
    pressure_advance = 0
    raft_layers = 0
    resolution = 0.01
    retract_before_travel = 0,2
    retract_layer_change = 1,1
    retract_length = 1,2.5
    retract_length_toolchange = 5,30
    retract_lift = 0,0.5
    retract_restart_extra = 0,0.5
    retract_restart_extra_toolchange = 5,10
    retract_speed = 20,20
    seam_position = random
    skirt_distance = 6
    skirt_height = 130
    skirts = 3
    slowdown_below_layer_time = 23
    small_perimeter_speed = 50%
    solid_infill_below_area = 70
    solid_infill_every_layers = 0
    solid_infill_extruder = 2
    solid_infill_extrusion_width = 0.38
    solid_infill_speed = 50%
    spiral_vase = 0
    standby_temperature_delta = -35
    start_gcode = G28 ; home all axes\nG1 Z30 F5000 ; lift nozzle\nG1 X100 F10800 ; \nT0\nM104 S[first_layer_temperature_0]\nM117 Move extruder until material emits from nozzle, then resume.\nM25\nM104 S195\nT1\nM104 S[first_layer_temperature_0]\nM117 Move extruder until material emits from nozzle, then resume.\nM25\nM104 S195\nG28 ; home again\nG1 Z5.0 F5000
    support_material = 1
    support_material_angle = 0
    support_material_enforce_layers = 2
    support_material_extruder = 1
    support_material_extrusion_width = 0.6
    support_material_interface_extruder = 1
    support_material_interface_layers = 2
    support_material_interface_spacing = 0
    support_material_interface_speed = 25%
    support_material_pattern = honeycomb
    support_material_spacing = 2
    support_material_speed = 80
    support_material_threshold = 0
    temperature = 230,230
    thin_walls = 1
    threads = 3
    toolchange_gcode = 
    top_infill_extrusion_width = 0.3
    top_solid_infill_speed = 25%
    top_solid_layers = 3
    travel_speed = 180
    use_firmware_retraction = 0
    use_relative_e_distances = 0
    use_volumetric_e = 0
    vibration_limit = 0
    wipe = 0,0
    xy_size_compensation = 0
    z_offset = 0

sirkha, 2015-01-16:
First two layers of G-Code 
===============
This illustrates that the skirt is appearing where it doesn't on the toolpath preview.
    ; generated by Slic3r 1.2.5 on 2015-01-15 at 18:44:52

    ; external perimeters extrusion width = 0.24mm
    ; perimeters extrusion width = 0.38mm
    ; infill extrusion width = 0.60mm
    ; solid infill extrusion width = 0.38mm
    ; top infill extrusion width = 0.30mm
    ; support material extrusion width = 0.60mm

    M107
    M190 S84 ; set bed temperature
    G28 ; home all axes
    G1 Z30 F5000 ; lift nozzle
    G1 X100 F10800 ; 
    T0
    M104 S230
    M117 Move extruder until material emits from nozzle, then resume.
    M25
    M104 S195
    T1
    M104 S230
    M117 Move extruder until material emits from nozzle, then resume.
    M25
    M104 S195
    G28 ; home again
    G1 Z5.0 F5000
    G21 ; set units to millimeters
    G90 ; use absolute coordinates
    M82 ; use absolute distances for extrusion
    G92 E0
    T0
    G92 E0
    M109 S230 ; wait for temperature to be reached
    G1 E-1.00000 F1200.00000
    G92 E0
    G1 Z0.300 F10800.000
    G1 X124.239 Y115.505 F10800.000
    G1 E1.00000 F1200.00000
    G1 X126.014 Y113.876 E1.04767 F2400.000
    G1 X129.111 Y112.747 E1.11288
    G1 X132.541 Y112.327 E1.18125
    G1 X133.430 Y112.273 E1.19888
    G1 X166.356 Y112.273 E1.85034
    G1 X167.194 Y112.321 E1.86695
    G1 X170.838 Y112.741 E1.93952 F2400.000
    G1 X173.153 Y113.409 E1.98719
    G1 X175.126 Y114.792 E2.03486
    G1 X176.541 Y116.743 E2.08253
    G1 X177.307 Y120.000 E2.14874
    G1 X177.307 Y180.000 E3.33586
    G1 X176.910 Y182.376 E3.38353
    G1 X175.761 Y184.495 E3.43120
    G1 X173.986 Y186.124 E3.47887
    G1 X170.889 Y187.253 E3.54408
    G1 X167.459 Y187.673 E3.61245
    G1 X166.570 Y187.727 E3.63008
    G1 X133.593 Y187.727 E4.28255
    G1 X132.743 Y187.678 E4.29939
    G1 X129.150 Y187.257 E4.37096
    G1 X126.836 Y186.586 E4.41863
    G1 X124.866 Y185.199 E4.46630
    G1 X123.454 Y183.247 E4.51397
    G1 X122.693 Y180.000 E4.57995
    G1 X122.693 Y120.000 E5.76707
    G1 X123.090 Y117.624 E5.81474
    G1 X124.214 Y115.552 E5.86138
    G1 E4.86138 F1200.00000
[... 444 lines cut ...]
    G1 X143.481 Y120.700 E1.01575 F4800.000
    G1 X140.095 Y120.700 E1.03820
    G1 X138.907 Y122.757 E1.05394
    G1 X135.521 Y122.757 E1.07639
    G1 X134.484 Y120.961 E1.09015
    G1 X134.484 Y123.156 E1.10470
    G1 X165.516 Y123.156 E1.31044
    G1 X165.516 Y122.757 E1.31309
    G1 X162.963 Y122.757 E1.33001
    G1 X161.776 Y120.700 E1.34576
    G1 X158.389 Y120.700 E1.36821
    G1 X157.202 Y122.757 E1.38396
    G1 X153.816 Y122.757 E1.40641
    G1 X152.628 Y120.700 E1.42215
    G1 X149.242 Y120.700 E1.44460
    G1 X148.055 Y122.757 E1.46035
    G1 X144.668 Y122.757 E1.48280
    G1 X166.100 Y123.740 F10800.000
    G1 X166.100 Y119.600 E1.51025 F4800.000
    G1 X133.900 Y119.600 E1.72374
    G1 X133.900 Y123.740 E1.75118
    G1 X166.100 Y123.740 E1.96467
    G1 E0.96467 F1200.00000
    G92 E0
    G1 Z0.450 F10800.000

lordofhyphens, 2017-03-19:
Looks alright to me. 
[image]

I'm not seeing weird gcode output either from 1.3.0-dev.

Closing; please provide updated test case if you can't replicate this (http://dl.slic3r.org/dev), attaching files as per https://help.github.com/articles/file-attachments-on-issues-and-pull-requests/

closed 2017-03-19 by lordofhyphens